Hypfer/Valetudo #421: startCleaningZoneByCoords "only" supports up to 5 zones

jonathan-reisdorf, 2020-02-01:
## Describe the bug
Hi @Hypfer ! 👋
Is there a reason behind startCleaningZoneByCoords "only" supporting up to 5 areas, e.g. does the vacuum stop handling things correctly at some point?
https://github.com/Hypfer/Valetudo/blob/9fc8db27d757b196b3036028468ec5f905d216f8/lib/miio/Vacuum.js#L628

I'm asking because some of my zones consist of multiple areas and when cleaning multiple zones at once via MQTT this limit is reached quite quickly. E.g. my floor zone consists of 3 areas and when I want to clean it twice in a row (it's possible to a zone more than once in the array and then it cleans twice, thanks for that, very handy! :smile:) it does not work.

## To Reproduce
Run the MQTT `zoned_cleanup` command with a total of more than 5 areas.

## Expected behavior
If there is a technical limitation, e.g. on the robot, a comment in the code about that might be nice, else maybe let's increase the limit? :slightly_smiling_face:

jonathan-reisdorf, 2020-02-01:
Nevermind, just saw your comment: https://github.com/Hypfer/Valetudo/issues/296#issuecomment-515970208

Still maybe a comment in the code would be great! :)

closed 2020-02-01 by jonathan-reisdorf

Hypfer, 2020-02-01:
A comment would indeed be helpful there.

Sadly, I cannot tell you anything about newer firmwares. I stopped at 1886