Hypfer/Valetudo #422: RFC: The future of Valetudo 2020+

Hypfer, 2020-02-01:
Moving forward, I guess that it is safe to say that there will be more vacuums at some point.
There's a Viomi one in #420 but that will certainly not be the last one.

This seems to be a good time to emphasize the mission of Valetudo.

### What Valetudo _is_
Valetudo aims to be a generic thing that you can put on your vacuum to remove cloud connetivity, provide a local MQTT Interface and maybe the Web interface as well, although I'm unsure about that. It might make sense to decouple it completely.

Valetudo should completely remove any vendor-specific details such as Map formats, Command formats etc.
Basically you should be able to buy a robot, install Valetudo and it works exactly the same as your existing Valetudo-enabled vacuum even if it is made by another vendor.

### What Valetudo _isn't_ 
Valetudo shall not become a swiss army knife of some sort. That means that everything that can be done elsewhere shall be done elsewhere. (e.g. connectivity to other services such as Telegram)

### How do we get there?
I think that Valetudo requires a major rewrite. The current state is basically just a prototype.

The main ToDos that come to my mind are the following:
#### Capabilities
This should feature some kind of capability system similar to the [supported_features](https://www.home-assistant.io/integrations/vacuum.mqtt/#supported_features) of an mqtt.vacuum in home assistant. Especially with the new room detection features, there's just no point in hacking those into the current codebase.

We'd need to figure out what those might be (preferably future-proof) and how to incorporate limits to it such as the 5-Zone Zoned Cleanup limit of the roborock vacuums which often caused confusion e.g. #421 

#### Map Data Format
There needs to be a proper Valetudo Map format which is future-proof and may contain new capabilities we don't even know yet. It will most likely be some binary blob since the current state using JSON just isn't feasible for larger setups as well as mobile access (#356)

Something like Protobuf could probably be used here.

#### Deployment
Valetudo binaries are _huge_ which worked out okay-ish at first but since vendors have now started trimming down hardware specs of their vacuum, I think it's safe to say that the current state isn't future-proof.

We're also using an old and deprecated version of the `pkg` tool used to package valetudo into a single binary since newer pkg versions require a newer libc. That will also certainly break completely at some point.

### Challenges
As you might have noticed, development of Valetudo slowed down quite a bit. There are multiple reasons for that but for the most part it's the fact that it is [mostly just me who worked on Valetudo](https://github.com/Hypfer/Valetudo/graphs/contributors).

Looking at Heise, private Messages, the comments in #400 and many more things, It became clear that the way I mentioned contributers apparently was misleading. **There is almost no community.**

There only thing there is plenty of are entitled end-users who don't understand FOSS _at all_ and just demand that things shall be done for them because reasons.

This ultimately led me to pretty much abandon the project because it just isn't fun.

There are also still unsolved issues with randomly resetting robots (#206) which constantly leads to a lot of frustration on all sides.

Also also there seem to be some interpersonal conflicts which I don't really understand but am unable to clarify since there's just silence.

### Moving forward
There however is a solution to the mentioned challenges: 

If you are unhappy with Valetudo but unable to contribute something to change that just use the non-fork fork [Valetudo-RE](https://github.com/rand256/valetudo) and get the fuck off my lawn.

I am so done with you people.

### Closing remarks

So anyways these are my plans for Valetudo.
As you may have noticed, this Issue is labeled RFC so feel free to add your thoughts

legomind, 2020-02-01:
@Hypfer I totally get your frustrations regarding the lack of contributors & community around the project. I happen to be one of the users dealing with the reset issue, and I regret that I am unable to add to the discussion regarding a solution. Thank you for being up front about the future of this project. Hopefully, the possibility of extending this project to other vacuums will entice more contributors. Your work on this has been amazing, I am sorry that there are many who do not seem to appreciate this project is FREE.

rumpeltux, 2020-02-03:
Thanks for this overview! A few drive-by comments:

* Mission: abstraction of robot specific API into a common one seams great. Things like MQTT / Web Interface could be optional modules or even separate binaries. This would also help clarify the abstraction API.
* Capabilities: totally makes sense
* Map Data Format: I don’t think json is necessarily so bad, in particular because it’s super easy on the client to handle (and thus also fast to parse) and the compressed size overhead compared to protobuf is probably <30% or so. However, I’m sure things can be optimized, probably a bit on the representation level and on the map query interval. E.g. send the bitmap at most once 30s or (this would mean about 10-20MB per hour if live-watching) and send incremental path-updates in between.
* Deployment: is the size limitation actually a problem today? What’s the limit / What will be the limit? I’d say keep resources in mind and don’t waste them, but do the same with the scarce developer resources, i.e. put them where the most benefit is.

alexkn, 2020-02-04:
First of all, thank you for your work.

In my opinion Valetudo is in a "works for me" status.
I just don't need any new features.
So I don't really know what to contribute.
And I think so do many Valetudo users who could also code.

Bigger wishes, like the support of other vacuum cleaners and new firmware versions are not really tasks for beginners.
New vacuum cleaners are also associated with spending money without knowing whether you can get it working.

Some ideas for moving forward:
I think the project should be splitted:
1) Code which really have to run on the vacuum which exposes a MQTT and/or REST API.
2) Webinterface
3) Supporting libraries, like the map parsing stuff, which could also be used in [ICantBelieveItsNotValetudo](https://github.com/Hypfer/ICantBelieveItsNotValetudo) or in my [node-red-contrib-valetudo](https://github.com/alexkn/node-red-contrib-valetudo) project.

If you then have the choice of what is actually installed on the vacuum cleaner, this could also solve the deployment problem.

matthiasharrer, 2020-02-06:
I really appreciate your work so far and I like that you are thinking about the future of the project and proposing some new ideas.

I can totally understand the frustration regarding users who just expect things to be done, opening duplicate issues, asking for help without reading the existing issues / wiki etc.
However I also think it is quite an achievement to have so many people interested in the project.

Maybe it would be an option to find someone who wants to help (and does not need to code) to manage the support issues etc.
This would free up the time for you and other contributors to work on things that they enjoy doing in their free time.

Although I am not sure how much time I could put into the project right now I would totally enjoy some additional communication channel which is more focused on the technical details going forward than the existing telegram channel. Hopefully some potential contributors would come together starting to discuss the technical bits, like splitting the project, rewriting bits etc in more detail.

alexkn, 2020-02-07:
@matthiasharrer 
Technical development questions can be discussed in github issues.
For helping normal users (Support requests) it would me nice to have a forum.
I don't like telegram for discussing anything.

JPT77, 2020-02-11:
Hi, I am a new Roborock user. 
Thanks for this project! It's great that we can make this robot work without the "cloud".
But the current state seems to be far from stable. I currently am frustrated and now considering just using manual control without any wifi connection.
I would like to help, but I don't know how much time I can invest. I am used to u-boot and building for arm. Fiddling with linux... and I am Java developer. So I would be OK with writing JS I think...or with any debugging tasks...
So yeah, probably we should develop a new architecture first and then distribute separate jobs to people.

kschuerholt, 2020-02-11:
Let me also second that and thank you for the great work you've put in already. I absolutely understand your frustration, any kind of demand for support or features is ludicrous.

The roadmap you've laid out seems sound. If there's a way you see to split further development into smaller, digestible chunks, people will consider developing it? I for once don't know if I could contribute on a large scale for lack of experience, but if a sub-project has a clear task and interfaces, I might be able to help.

matthiasharrer, 2020-02-12:
@alexkn You are probably right. We should setup a support forum and find some people willing to contribute as form mods .. this should hopefully also move support questions from github issues over to the forum.

Maybe we should also introduce a new label for marking issues for the rewrite/rearchitecture, so that they are easy to spot.

rumpeltux, 2020-02-12:
> @alexkn You are probably right. We should setup a support forum and find some people willing to contribute as form mods .. this should hopefully also move support questions from github issues over to the forum.

Maybe it makes sense to ask the folks from roboter-forum.com to create a sub-forum for valetudo. It’s already being discussed there quite extensively.

matthiasharrer, 2020-02-12:
@rumpeltux I also thought of this and like the idea, however roboter-forum.com seems to be mostly in german. It's a good idea nonetheless as I think most of the users are from germany currently.

balu-, 2020-02-24:
hi @Hypfer 
thanks for making this project happen.
Also thanks for laying out your thoughs so clearly as you did.
I totaly get your frustration, about the "one man community" and the "demanding" attitude of some users. Further I'd totaly secound your view on the rewrite step that needs to be done to get to a "bright valetudo future".
But at least in my opinion a very big part of the current situation isn't pure codequality.
For me the Gordian knot is #206 (the random firmware resets) which i think needs a solution first.

**why do i think that is such a big deal**

- it annoys everyone who has to reflash her/his robot. 
(Especialy because the time that i notice i need to reflash is most often when I don't have it)
Further I have noticed a few comments that were along the lines "to much effort i use my robo the normal way" - which is realy sad in my opinion (I would argue that those aren't "entitled end-users" but more "helpless users")  

- it's not clear if this issue is solvable.
Correct me if I'm wrong but as far as I know noboy really knows exactly where this issue comes from (it seems like its something the firmware does but it's uncertain if it is triggerd by rooting or using valetudo or if its just not recognized in normal use). So it might never be solvable which seems like a real bummer & doesn't motivate me to put time in the project (as my skills are not enough to fix this issue).

- it makes me unable to recomend the project to most people.
I would love to tell all my nerd friends "buy a roborock and run valetudo - its cloudless awesomness", unfortunately I can't. I would have to explain the whole reflash story - and feel bad if they run into this issue - so I just don't tell

I know those are my very personal reasons, but I imagine that those are reasons that are at least in part holding the community back.

I really hope we find a solution to this issue. 
[If there is anything i can do for that please let me know]

deg0nz, 2020-03-01:
I have to say, I can understand your frustration and feel with you. This is not easy. I know this from 1-2 repos I have publicly available myself. So I want to thank you for this and say to keep up the work. It is really appreciated here. You saved me a lot of work with this...
For me, Valetudo is also in a "works for me" status, as @alexkn said. So no complains.

I have a question about the rewrite:
Are you planning to leave it in plain Javascript or are you considering switching to TypeScript?
Or even a completely different language (as Rust for example)? (This could solve your space-problem)

When I have finished my current project, and I find time, I'm happy to contribute to your rewrite. I'll watch this issue to check the progress now and then.

Hypfer, 2020-03-01:
@deg0nz 
Thank you. Oh and thanks to everyone else who commented as well ❤️ 

See #426 for the rewrite.

Personally, I don't really like Typescript since it adds additional complexity without any meaningful reward. That _might_ have changed, but so has ES6.

We definitely need to solve the deployment issues (outdated and broken zeit/pkg, less storage & memory on newer vacuums) in the foreseeable future so using something like rust could definitely make sense. Only problem being that I couldn't contribute much to that codebase :(

I guess it should be doable using JS. We'd just need to figure out how to get the runtime on the robot. I haven't looked into that yet. zeit/pkg was choosen at the time because it was easy and self-contained

t-animal, 2020-03-05:
Honestly the decision not to switch to typescript has stopped me personally from contributing. I asked whether this would be wanted in the telegram group some time ago and it was shunned upon. Having developed javascript some time and now only typescript in my professional life I never wanna go back, certainly not in my free time. This interaction even led me to giving talks about typescript now, because I always thought the switch was a no-brainer.

This having been said, I think you're absolutely on track with the design decisions mentioned above. Decouple the frontend and the software on the robot. Let others do frontends how they like, be it react or vue or plain HTML and some JS (modern, hopefully), be it typescript or javascript. Be it a website, PWA or native app. Offer a set of APIs. This will not only make your life easier but also hopefully relief you of the burden to have to answer users' complaints in issues. A prerequisite would probably be to find someone to develop at least one frontend project.

Lastly I just wanna say thank you. My robot resets every couple of weeks and I have thought multiple times about going back to only using flolevac. But still, I flash valetudo every time, because it just adds so much usefulness. Just did it again today and probably will continue. Thanks.

HHenning0815, 2020-04-10:
I see your frustration, but...

Just close a support issue without notice why ists closed (without chance that someone can answer) is also frustrating.

I know that I have no clue about programming or so, but all that I hoped for is a little help to know what I can do to fix my problems.

rumpeltux, 2020-04-13:
> I see your frustration, but...
> 
> Just close a support issue without notice why ists closed (without chance that someone can answer) is also frustrating.
> 
> I know that I have no clue about programming or so, but all that I hoped for is a little help to know what I can do to fix my problems.

@HHenning0815: I think you’re referring to #473. I hope you can understand that we cannot provide individual support for issues not directly related to the Valetudo software. That said, I do agree, that bugs (even *support requests*) shouldn’t be closed without comments.
We should setup a canned response that points people to a better place to ask such questions (e.g. for now, in lieu of a better alternative this could be roboter-forum.com).

docBliny, 2020-05-11:
@Hypfer While this isn't the correct forum for this, I just wanted to drop by and say thanks for the work on this project! I just updated our two gen1s (and hopefully the reset fix sticks this time) and it reminded me to post.

Hypfer, 2020-06-01:
Regarding the map data format:

It might make sense to rethink this one. The whole concept is heavily biased by how roborock saw things but that's not necessarily the right approach for a generic abstraction like Valetudo.

Currently, it contains both "static" as well as "dynamic" state data.

"Dynamic" state data includes things such as:
- Robot Position
- Go-To Target
- Predicted Go-To Route
- Currently cleaned zones
- Path information

while "static" Map data includes these:
- Map Pixels
- Segment Pixels
- Segment Names (?)
- Virtual Walls/ No-go Areas
- Charger Position
- Floor Name/Id/???

Basically there are these two very different questions one might ask the robot:
- What can you tell me about yourself?
- What can you tell me about your environment?

Which data source is used in the background to answer these questions should be irrelevant.

combative-american, 2020-06-18:
Wasn't really sure where to reach out so I figured I would try here. I have an E35 vacuum. The readme was specific that only the older models are supported (Mine is May2019). Will you be adding the ability / readme / video how-to to root these newer vacuums? Even it it requires opening the vacuum (as long as I dont have to solder anything).

Hypfer, 2020-06-18:
You cannot run Valetudo on an E35 since it doesn't even run linux at all.

You could certainly try dumping whatever storage there might be to acquire the CloudKey as well as the DID, but there hasn't been any research on those budget models yet.

Furthermore, I don't own one and in fact I'm not even the one who did and does all the root magic.

And, lastly, this issue isn't even remotely the right place for this question.

closed 2020-07-31 by Hypfer