Hypfer/Valetudo #426: Modernize Infrastructure: Support ES6 / Typescript

rumpeltux, 2020-02-12:
imho it would be beneficial for the project to modernize the infrastructure so that new code can be written in typescript or at least ES6.

This has a bunch of benefits like:
* having proper support in IDEs, whose automated error checking can avoid a whole class of bugs
* being able to more easily define interfaces and class hierarchies
* potentially being able to compile the code to a more efficient representation which may alleviate some of the binary-size concerns

Similarly the project should probably adopt a linter. I’m not sure what the state of the art is for either though.

rumpeltux, 2020-02-12:
That is assuming that the new architecture is still based on nodejs (which I think does make sense given the existing codebase and the easiness of the development cycle).

alexkn, 2020-02-12:
I am unsure whether we should use Typescript or ES6.
@rumpeltux  What are the advantages of Typescript compared to ES6 with activated type checking?
If there are no real advantages, I would want to avoid the additional compilation of Typescript.

I use ESLint in some other projects.
I also created GitHub Actions there to automatically test commits and pull requests with the Linter.

rumpeltux, 2020-02-12:
I found https://stxnext.com/blog/2019/08/30/typescript-pros-cons-javascript/
tl;dr benefits are minimal once you consequently use jsdoc annotations, examples less verbose type writing, compile-time errors

So sticking with jsdoc makes sense. ES6 classes seem to work. ES6 imports don't, but I guess I can live with that.

@alexkn Could you set up eslint for the project?

alexkn, 2020-02-12:
> @alexkn Could you set up eslint for the project?
Yes, I will do that.

alexkn, 2020-04-05:
@rumpeltux Could you refactor the ViomiMapParser to a class?
Then we can close this issue.

rumpeltux, 2020-04-13:
> @rumpeltux Could you refactor the ViomiMapParser to a class?
> Then we can close this issue.

Done :-)

closed 2020-04-13 by rumpeltux