Hypfer/Valetudo #428: Separate Robot Interface from other components

rumpeltux, 2020-02-13:
I created https://github.com/Hypfer/Valetudo/wiki/New-Vacuum-Interface to discuss the interface we want / need. Please extend, discuss and chime in.

This is part of the work on #422.

alexkn, 2020-05-14:
The document was moved to https://github.com/Hypfer/Valetudo/blob/master/docs/new-vacuum-interface.md

But I don't think we should go that theoretical way, because we don't have the manpower to do a full rewrite.

By now we have a kind of abstract interface with the MiioVacuum class.
I think we should use that as a base and refactor it in a more organic way when features for other vacuums like viomi are implemented.
We can extract a fully abstract Vacuum class when a non-miio vacuum is implemented.

What do you think @Hypfer @rumpeltux ?

Hypfer, 2020-05-15:
>premature optimization is the root of all evil

I do agree that we should definitely see more vacuum implementations before we build a proper abstraction, however I think that we could probably already enhance the Map and Status data formats.

Maybe we should move those into seperate issues?

Hypfer, 2021-01-09:
Done

closed 2021-01-09 by Hypfer