Hypfer/Valetudo #429: Create a test suite and integrate the tests into CI jobs

rimrul, 2020-02-14:
As part of the rework we should introduce a way to automatically test changes and reduce the chances of introducing regressions.

rimrul, 2020-02-14:
There was some discussion about this in #427 and it got a little off-topic forr that issue.

Testing on armv7 trusty probably makes sense to get close to testing on an actual roborock. I'll investigate if we could get a trusty-armv7 docker image going on github actions (using [uraimo/run-on-arch-action](https://github.com/uraimo/run-on-arch-action) and a custom image) since alex is introducing Github actions CI with #427 anyways and they apparently continue running on forks without reconfiguration.

rimrul, 2020-02-14:
@rumpeltux could you lookup the musl libc and kernel versions on your Viomi, so we can try to get that image close to what the actual robot offers?

rumpeltux, 2020-02-14:
> @rumpeltux could you lookup the musl libc and kernel versions on your Viomi, so we can try to get that image close to what the actual robot offers?

musl is 1.1.14 and kernel is 3.4.39

If it’s of any help I have a working toolchain to build the tina linux kernel and base image ([documented here](https://itooktheredpill.irgendwo.org/2020/rooting-xiaomi-vacuum-robot/#3-installing-ssh)). But I guess a close enough match is fine (I tried a random alpine-linux-armv6 nodejs binary and it seemed to work, see [viomi/deploying](https://github.com/Hypfer/Valetudo/blob/viomi/viomi_instructions.md#deploying))

rimrul, 2020-02-14:
Yeah, I think close enough is fine atm. Just testing against the latest amd64 glibc and kernel felt a little to far off. We can dial it in closer if we feel like we need to, but compiling kernels sounds a little overkill for now. Thank you.

rimrul, 2020-03-07:
I have the first compiling CI build on armv7-trusty (Node 9).

closed 2021-03-23 by Hypfer