Hypfer/Valetudo #431: [viomi branch] Local communication breaks once the robot connects to Dummycloud

rumpeltux, 2020-02-15:
Per report by @Hypfer in https://github.com/Hypfer/Valetudo/pull/420#issuecomment-586488471.

rumpeltux, 2020-02-15:
That’s weird, the states are actually isolated and shouldn’t affect each other. They also don’t do so for me and when using the fakerobot implementation to test.
One thing I did change, was to add [ping/pong handling to miiosockets](https://github.com/Hypfer/Valetudo/blob/viomi/lib/miio/MiioSocket.js#L60). Once the cloud connects and we respond with a pong, it could be that the robot disables the local interface? I think that’s somewhat unlikely though. In any case, a workaround would be to send all (even the local) commands through the cloud interface once it’s available.

Hypfer, 2020-02-15:
I'd like to ditch the local interface anyways. The only reason why we need it is to provision the robot and save the wifi credentials

rumpeltux, 2020-02-15:
There’s the argument to be made that a bunch of functionality of Valetudo works without the cloud interface, e.g. you could run Valetudo on your raspi or so without having to root the robot at all.

rumpeltux, 2020-02-15:
60610e5 should address this (though I wasn’t able to repro the original bug).

There’s one caveat: the cloud connection doesn’t have fancy retry logic implemented, which is probably not a big deal since udp connections are reliable when on the same machine.

Hypfer, 2020-02-15:
Sadly no. On a fresh boot it still gets stuck in a timeout loop for the local interface. :(
It does seem to be a timing issue. Adding a 10s sleep to the upstart file does seem to avoid this.

I do like how you've transparently rerouted the local communication to the cloud. Thats a lot simpler and easier than I thought it would be.

rumpeltux, 2020-02-16:
I just stumbled upon #259 which sounds somewhat similar.

When you say that this continues to happen even after switching to cloud communication I’m confused, because no more messages should be send on the local interface after doing so.
The initial local handshake is succeeding though?

I guess a 10s delay could be acceptable. Would it make sense to start valetudo after the robot binaries (e.g. in `/etc/rc.d/S97valetudo` or so) and would this be sufficient to avoid the problem?

Hypfer, 2020-02-16:
Actually it's the same request message getting retried. New requests are indeed routed through the cloud interface and therefore work. 

I'm never seeing any handshake success messages. In fact it doesn't ever seem to even enter RetryWrapper.onMessage at all.

Why is onMessage null?
https://github.com/Hypfer/Valetudo/blob/viomi/lib/devices/MiioVacuum.js#L57

`/etc/rc.d` is not available on roborock vacuums. We could however probably use `/opt/rockrobo/watchdog/rrwatchdoge.conf` to let the roborock tooling handle it.

Hypfer, 2020-02-16:
Increasing the miio_client logLevel reveals this:
```
[20200216 15:17:50] [DEBUG] look like a repeated msg:-2,1
[20200216 15:17:50] [INFO] OT protocol diagrams come...
[20200216 15:17:50] [DEBUG] look like a repeated msg:-2,1002
[20200216 15:17:51] [INFO] OT protocol diagrams come...
[20200216 15:17:51] [DEBUG] look like a repeated msg:-2,2003
[...]
[20200216 15:18:05] [INFO] OT protocol diagrams come...
[20200216 15:18:05] [DEBUG] look like a repeated msg:-2,30031
```

Valetudo Log (with extra console logs)
```
STAMP Stamp { val: 1581866220, time: 1581866220116 }
[2020-02-16T15:17:00.216Z] [DEBUG] >>> local: {"method":"get_status","params":{},"id":1}
[2020-02-16T15:17:00.718Z] [DEBUG] local request 1 timed out
STAMP Stamp { val: 1581866220, time: 1581866220116 }
[2020-02-16T15:17:01.222Z] [DEBUG] >>> local: {"method":"get_status","params":{},"id":2003}
[2020-02-16T15:17:01.723Z] [DEBUG] local request 2003 timed out
[...]
STAMP Stamp { val: 1581866220, time: 1581866220116 }
[2020-02-16T15:17:15.291Z] [DEBUG] >>> local: {"method":"get_status","params":{},"id":30031}
[2020-02-16T15:17:15.792Z] [DEBUG] local request 30031 timed out
```

Meanwhile using master + extra console logs, there are almost no retries

```
RETRY No 1 Stamp { val: 869, time: 1581866965481 } 1
RETRY No 2 Stamp { val: 869, time: 1581866965481 } 1001
RETRY No 1 Stamp { val: 869, time: 1581866965747 } 2
RETRY No 2 Stamp { val: 869, time: 1581866965747 } 1002
RETRY No 1 Stamp { val: 874, time: 1581866970872 } 3
RETRY No 2 Stamp { val: 874, time: 1581866970872 } 1003
RETRY No 3 Stamp { val: 874, time: 1581866970872 } 2003
RETRY No 1 Stamp { val: 879, time: 1581866976135 } 4
```

I did wonder why the stamp looks this different. Which lead me to https://github.com/Hypfer/Valetudo/blob/viomi/lib/miio/MiioSocket.js#L66

Something definitely got mixed up there while refactoring. Local communication must just re-use the stamp as received from miio_client and not change anything about it iirc.

Also, I've noticed that the retry wrapper doesn't work at all since `options.retries` is always undefined in onTimeout

rumpeltux, 2020-02-16:
Thanks. I didn’t realize that codec is updating the outgoing stamp and the logic is a bit obscure. Hopefully it’s better now. Please retry.

re options.retries. in the first onTimeout it’s undefined, but subsequently it should be increased. And throw an exception after 10000 tries.
I don’t think I changed the logic compared to:
https://github.com/Hypfer/Valetudo/blob/master/lib/miio/Vacuum.js#L103

The exiting due to exception didn’t work anymore though because exceptions within promises are swallowed and don’t exit anymore. Updated the code.

Hypfer, 2020-02-16:
Now it enters a different loop and much faster loop:

```
[2020-02-16T16:58:20.357Z] [DEBUG] >>> local: HandshakePacket()
[2020-02-16T16:58:20.382Z] [INFO] mqtt start
[2020-02-16T16:58:20.408Z] [INFO] Dummycloud is spoofing 203.0.113.1:8053 on 0.0.0.0:8053
[2020-02-16T16:58:20.409Z] [INFO] Webserver running on port 3197
[2020-02-16T16:58:20.508Z] [DEBUG] === local: still waiting for handshake
[2020-02-16T16:58:20.509Z] [DEBUG] >>> local: HandshakePacket()
[2020-02-16T16:58:20.660Z] [DEBUG] === local: still waiting for handshake
[2020-02-16T16:58:20.661Z] [DEBUG] >>> local: HandshakePacket()
[2020-02-16T16:58:20.812Z] [DEBUG] === local: still waiting for handshake
[2020-02-16T16:58:20.813Z] [DEBUG] >>> local: HandshakePacket()
[2020-02-16T16:58:20.963Z] [DEBUG] === local: still waiting for handshake
[2020-02-16T16:58:20.964Z] [DEBUG] >>> local: HandshakePacket()
[2020-02-16T16:58:21.114Z] [DEBUG] === local: still waiting for handshake
[2020-02-16T16:58:21.115Z] [DEBUG] >>> local: HandshakePacket()
[2020-02-16T16:58:21.266Z] [DEBUG] === local: still waiting for handshake
[2020-02-16T16:58:21.267Z] [DEBUG] >>> local: HandshakePacket()
[2020-02-16T16:58:21.418Z] [DEBUG] === local: still waiting for handshake
[2020-02-16T16:58:21.419Z] [DEBUG] >>> local: HandshakePacket()
[2020-02-16T16:58:21.570Z] [DEBUG] === local: still waiting for handshake
[2020-02-16T16:58:21.571Z] [DEBUG] >>> local: HandshakePacket()
[2020-02-16T16:58:21.722Z] [DEBUG] === local: still waiting for handshake
```

rumpeltux, 2020-02-16:
This seems to be a connection issue rather. The handshake parts were actually not touched in the recent commits.
I have this too sometimes if the robot is "slow" or wifi is crowded and udp packets get dropped. But it’s usually resolved after 3-10 tries.
Does this continue forever? Then the robot is probably not receiving the packets (e.g. wrong port / ip, firewall…)
The timeout of 100ms is quite aggressive though I guess.

Hypfer, 2020-02-16:
It's definitely a logic error since it works reliably with the current master branch.

I'll look into it tomorrow

Hypfer, 2020-02-17:
Fixed with e8dec559

closed 2020-02-17 by Hypfer