Hypfer/Valetudo #434: Info pane not loading

stefanct, 2020-02-16:
## Describe the bug

This has been bugging me for some time (where some time is about 1,5 years) and it survived a few updates already: on one of two gen2 roborocks I cannot get the info page to load. The spinner keeps spinning for a very long time (not sure if it ever times out actually).

Everything else works fine and has been in the past. Unfortunately, the valetudo.log does not shown any problems at all and there does not seem to be any CLI switches or env variables to enable more verbose output.

It is not that important but I would like to debug it anyway but have no idea how.
When I compare the two robots I can see the following:
 - both are currently running the same binary, built by myself some months ago based on 9fc8db2 from 2019-11.
 - both have this in their `/etc/OS_VERSION`

    ro.product.device=MI001518_RUBY-S_MP_S2018062700REL_M3.3.0_RELEASE_20180627-124454
    ro.build.display.id=RUBY-S_MP_R16_RELEASE_20180627-124454
    ro.sys.cputype=R16.STM32.A3.G1
    ro.build.version.release=001518
    ro.build.date.utc=1530074694

 - the iptables are equal (including the nat table)
 - the `valetudo/config.json` are very different (the one where it works has only an absolute minimal config) but I assume it's not related(?)
 - the hosts files are equal
 - the only difference that I think might matter is a line in the output of `netstat -4np`. The one that does work has this output:

    tcp        1      0 127.0.0.1:40256         127.0.0.1:80            CLOSE_WAIT  29473/AppProxy 

On the good one this connection (with another src port of course) is `ESTABLISHED`. All other connections seem to be equal-ish.
Any idea what could provoke the half-closed connection? AFAICT this would mean that valetudo has once established a connection with AppProxy but then closed it (while AppProxy has not closed it (yet)).

Hypfer, 2020-02-25:
You could try the latest master and set `logLevel` to `trace`.

Hypfer, 2020-04-12:
Any update on this @stefanct ?

stefanct, 2020-04-13:
No. I will look into it when I upgrade next time... but it works too well otherwise to bother tinkering with it ;)

Hypfer, 2020-05-15:
Please open a new issue when that happens

closed 2020-05-15 by Hypfer