Hypfer/Valetudo #438: [viomi branch] Valetudo shutdowns due to "Unable to reach vacuum"

Hypfer, 2020-02-19:
Using the latest viomi branch, on _some_ launches, valetudo restarts itself after ~1.5h uptime due to a retried message reaching the 10.000 retries limit.

I've observed this behaviour on two seperate S50.

Logs attached:

```
[2020-02-19T02:14:01.581Z] [INFO] Webserver running on port 80
[2020-02-19T02:14:06.704Z] [INFO] Connected successfully to mqtt server
[2020-02-19T02:14:09.584Z] [INFO] Cloud connected
[2020-02-19T03:37:33.152Z] [ERROR] Unable to reach vacuum
```
```
[2020-02-19T08:00:10.458Z] [INFO] Webserver running on port 80
[2020-02-19T08:00:16.592Z] [INFO] Connected successfully to mqtt server
[2020-02-19T08:00:18.452Z] [INFO] Cloud connected
[2020-02-19T09:23:41.852Z] [ERROR] Unable to reach vacuum
```

```
[2020-02-17T21:12:19.945Z] [INFO] Webserver running on port 80
[2020-02-17T21:12:33.358Z] [INFO] Cloud connected
[2020-02-17T22:35:51.252Z] [ERROR] Unable to reach vacuum
```

I've patched the code running on one S50 to report the offending message + current stamp information

```
[2020-02-18T07:21:43.124Z] [INFO] Webserver running on port 80
[2020-02-18T07:21:51.594Z] [INFO] Cloud connected
[2020-02-18T08:45:14.572Z] [ERROR] Unable to reach vacuum {
  "method": "get_status",
  "args": {},
  "options": {
    "retries": 10001
  },
  "stamp": {
    "val": 17,
    "time": 1582010503136
  }
}
```
```
[2020-02-19T02:42:42.004Z] [INFO] Webserver running on port 80
[2020-02-19T02:42:50.116Z] [INFO] Connected successfully to mqtt server
[2020-02-19T02:42:50.394Z] [INFO] Cloud connected
[2020-02-19T04:06:13.585Z] [ERROR] Unable to reach vacuum {
  "method": "get_status",
  "args": {},
  "options": {
    "retries": 10001
  },
  "stamp": {
    "val": 17,
    "time": 1582080162033
  }
}
```

In any case, 10.000 retries seem to be far too many. There's no point in receiving a response an hour later

alexkn, 2020-02-19:
I have this too:

```
[2020-02-19T03:06:40.410Z] [DEBUG] local request 19900883 timed out
[2020-02-19T03:06:40.411Z] [DEBUG] >>> local: {"method":"get_status","params":{},"id":19902885}
[2020-02-19T03:06:40.881Z] [DEBUG] local request 19901884 timed out
[2020-02-19T03:06:40.882Z] [DEBUG] >>> local: {"method":"get_consumable","params":[],"id":19903886}
[2020-02-19T03:06:40.913Z] [DEBUG] local request 19902885 timed out
[2020-02-19T03:06:40.916Z] [ERROR] Unable to reach vacuum
[2020-02-19T03:06:40.920Z] [INFO] exiting...
```

That `local request timed out` started hours before, several times per second.

alexkn, 2020-02-20:
I set this retries option from 10000 down to 5:
Now I had this in my log:
```
[2020-02-20T01:42:22.992Z] [INFO] Valetudo shutdown done
[2020-02-20T01:42:22.994Z] [INFO] exiting...
[2020-02-20T01:42:47.019Z] [INFO] Loading configuration file: /mnt/data/valetudo/config.json
[2020-02-20T01:42:47.053Z] [INFO] Starting Valetudo for Vacuum Model roborock.vacuum.s5
[2020-02-20T01:42:47.055Z] [INFO] DID xxxxxxxxxxx
[2020-02-20T01:42:47.056Z] [INFO] CloudKey xxxxxxxxxxx
[2020-02-20T01:42:47.098Z] [DEBUG] >>> local: HandshakePacket()
[2020-02-20T01:42:47.171Z] [INFO] mqtt start
[2020-02-20T01:42:47.367Z] [INFO] Dummycloud is spoofing 203.0.113.1:8053 on 127.0.0.1:8053
[2020-02-20T01:42:47.372Z] [INFO] Webserver running on port 80
[2020-02-20T01:42:47.383Z] [DEBUG] === local: still waiting for handshake
[2020-02-20T01:42:47.384Z] [DEBUG] >>> local: HandshakePacket()
[2020-02-20T01:42:47.399Z] [DEBUG] <<< local* {"stamp":17}
[2020-02-20T01:42:47.402Z] [DEBUG] <<< local* {"stamp":17}
[2020-02-20T01:42:47.404Z] [DEBUG] <<< local* {"stamp":17}
[2020-02-20T01:42:47.536Z] [DEBUG] === local: still waiting for handshake
[2020-02-20T01:42:47.552Z] [DEBUG] >>> local: {"method":"get_status","params":{},"id":1}
[2020-02-20T01:42:48.041Z] [DEBUG] local request 1 timed out
[2020-02-20T01:42:48.047Z] [DEBUG] >>> local: {"method":"get_status","params":{},"id":1002}
[2020-02-20T01:42:48.546Z] [DEBUG] local request 1002 timed out
[2020-02-20T01:42:48.549Z] [DEBUG] >>> local: {"method":"get_status","params":{},"id":2003}
[2020-02-20T01:42:49.049Z] [DEBUG] local request 2003 timed out
[2020-02-20T01:42:49.053Z] [DEBUG] >>> local: {"method":"get_status","params":{},"id":3004}
[2020-02-20T01:42:49.552Z] [DEBUG] local request 3004 timed out
[2020-02-20T01:42:49.557Z] [DEBUG] >>> local: {"method":"get_status","params":{},"id":4005}
[2020-02-20T01:42:50.055Z] [DEBUG] local request 4005 timed out
[2020-02-20T01:42:50.059Z] [DEBUG] >>> local: {"method":"get_status","params":{},"id":5006}
[2020-02-20T01:42:50.558Z] [DEBUG] local request 5006 timed out
[2020-02-20T01:42:50.561Z] [DEBUG] >>> local: {"method":"get_status","params":{},"id":6007}
[2020-02-20T01:42:51.060Z] [DEBUG] local request 6007 timed out
[2020-02-20T01:42:51.063Z] [ERROR] Unable to reach vacuum
[2020-02-20T01:42:51.065Z] [INFO] exiting...
```
Valetudo has just restarted normally before. I think that was the normal nightly reboot.

rumpeltux, 2020-02-23:
My theory is as follows:
on boot the device doesn’t have the current time, thus the timestamp is <100.
Soon after, the device updates local time either from filesystem or network. Now all the stamps we send won’t be accepted anymore, thus the requests time out.
I just pushed https://github.com/rumpeltux/Valetudo/tree/handshake which performs another handshake if a request timed out.
If my theory is correct this should solve the issue.

Hypfer, 2020-02-23:
@rumpeltux Yup. Fixed.

That should mark the last big regression, I think. It's time to merge viomi to master

alexkn, 2020-02-23:
Yes, looks good.

commit d90775c6 (2020-02-23): Agressively re-handshake. Trying to fix #438. (#443)
closed 2020-02-23 by Hypfer