Hypfer/Valetudo #439: Valetudo JSON Map Format can't handle larger maps

Hypfer, 2020-02-19:
Repeatedly parsing a larger map file inevitably leads to valetudo crashing due to it going out of memory. I've tried mitigating this by rewriting the parser in a non-recursive way in b6c67fcb but that sadly couldn't fix it. It did improve readability though.

While testing the new parser, I however found the underlying issue: Filling an array with thousands of 2-element arrays just isn't feasible at all.
Rewriting that part to use typed Int16 Arrays reduced the memory footprint of the test .js file from 40mb for a single map parsing to a mere 9mb.
Using this new format would of course break everything consuming the map data atm.

Since there should be a new and future-proof valetudo map format soon™, I don't think that breaking everything twice is a good idea, but I also strongly dislike the constant crashes I'm seeing.

For reference, here's a mapfile that breaks the parser if parsed repeatedly using the current parser and `--max-old-space-size=32`
[mapfile.zip](https://github.com/Hypfer/Valetudo/files/4227089/mapfile.zip)

alexkn, 2020-02-20:
I cannot reproduce your problem.
I don't see any memory problems on my S50.
I also tried to test the parser in isolation with your map file.
It runs for multiple minutes without any memory issues.

I don't like your typed Int16 Array idea.
The JSON representation of that look strange.

Maybe we shouldn't parse the whole image block.
Only the position and dimensions.
Then simply put the rest of the block as array into the JSON.

As far as I understand, there is simply one byte per pixel in the native buffer, and the respective value indicates the type of pixel.
The ViomiMapParser looks like it is the same there.
We can include a pixel type mapping in the JSON, so client can directly draw an image out of it.
That should also make the JSON smaller.

Hypfer, 2020-05-01:
Currently, the map uses pixels while paths and positions are in millimeters.

When desiging a new map format, we should definitely only use one way of specifying this data type.

Hypfer, 2020-06-04:
This looks okay with more recent nodejs version. Closing this issue for now

closed 2020-06-04 by Hypfer