Hypfer/Valetudo #441: FW disappears when vac. gets disconnected from Wifi?

NOTORIOUSVR, 2020-02-20:
Back in December I had an issue where it seemed like the VAC disconnected from my Wifi and I had to reload the FW to get it back - it was somewhat of an ordeal detailed here: https://github.com/dgiese/dustcloud/issues/254

I chalked it up to a random issue or maybe something I did wrong even though the vac. worked flawless for about 2 weeks before that point.  I do remember having a power outage in that date range as well just before it happened.

It has now happened again - but instead of going crazy I just reinstalled the Mi app and let it search for the vac.  it found it and I was able to reconnect to my Wifi,  The vac. showed my full map (had map saving on in your FW) but I wasn't able to access the vac. internal web GUI for Valetudo.

I haven't yet attempted to reinstall the FW maybe hoping that someone here can help me understand why this is happening.

My best guess of the commonalities is that both times there was an extended power outage that caused the Robot to be disconnected from the Wifi AP - when the AP came back (SSID was back online) the internet connection was NOT due to my server/router not coming back up when the AP did.

I think at this point something happens with the FW and it possibly reloads the stock FW, but I cannot be sure.

LocNgu, 2020-02-22:
I had this issue twice now, but I didn't expierenced any power outages or anything else. The Robot disconnects from my wifi and open up its own ap. GUI is not available over this ap. It's like Valetudo just disappered from on day to another.

Hypfer, 2020-02-22:
#206

closed 2020-02-22 by Hypfer