Hypfer/Valetudo #450: Firmware Installation fails

Riseryn, 2020-03-03:
## Describe the bug
Hello I have followed the installation instruction but when i want to upgrade the firmware i got this
```
 mirobo --ip 192.168.8.1 --token  68534b4339786276467936d35534a4d73 update-firmware --ip 192.168.1.12 firmware/output/v11_003468.fullos.pkg
Going to update from firmware/output/v11_003468.fullos.pkg
INFO:miio.updater:Serving on 0.0.0.0:34161, timeout 10
INFO:miio.updater:Using local firmware/output/v11_003468.fullos.pkg (md5: 41a584c7f65277d3a055255ecc3151b3)
Hosting file at http://192.168.1.12:34161/v11_003468.fullos.pkg
Update started!
  0%|                                                                                                | 0/100 [00:05<?, ?it/s]
ERROR:miio.updater:No request was made..

```
vacuum say try updating after the vacuum is at least 20% charged.

but it is fully charged

## To Reproduce

### Screenshots

### Vacuum Model
rockrobo gen 1

### Valetudo Version

### User-Agent

## Expected behavior

## Additional context

closed 2020-03-04 by Hypfer

ASdeWekker, 2020-04-08:
Hi, I'm also having this issue while trying to flash the firmware. I'm currently on version 3.3.9_001886.
What did you do to resolve this?

ualex73, 2020-04-09:
Unless you are a network export, the following is really wrong:
mirobo --ip 192.168.8.1 --token  774c7a35656e52544234465161776f75 update-firmware --ip 192.168.1.12 firmware/output/v11_003468.fullos.pkg

The roborock is using 192.168.8.1 which is a DIFFERENT subnet then 192.168.1.1.12.

The roborock can *NEVER* reach the 192.168.1.12 IP address, there is no default gateway on the roborock. The URL for the firmware *needs* to be in the 192.168.8.XXX range.