Hypfer/Valetudo #451: Turn of device if not cleaning or charging

drtorchwood, 2020-03-07:
Hi, 

I would propose the following improvement:
If the cleaner is not running and not connected to the charger it should turn of after some time. 

Background:
My charger is getting warm even when the cleaner is 100 % loaded which indicates some waste of energy. Therefore, I used an external timer to turn off the charger and turn it on shortly before a scheduled cleaning. 
This worked well with the original xiaomi software: after turning off the power supply, the robot turned off, too. Switching on the charger, made the cleaner to boot. 
With valetudo, however, the robot does not turn off.

Hypfer, 2020-03-07:
No.

closed 2020-03-07 by Hypfer

rumpeltux, 2020-03-07:
The robot implicitly shuts down when the battery is empty anyways.
You can ssh to the robot and issue the `poweroff` / `shutdown` command manually.

codingstube, 2021-03-28:
I can confirm, that a shutdown with valetudo does not work if the vacuum is not docked and battery is charged.
With original cloud. It shuts down after some time: 5 minutes or so.

If I poweroff via ssh the viomi robot restarts and not halt. :(