Hypfer/Valetudo #456: MQTT HA Discovery tweaks

LordMike, 2020-03-14:
Hi,

From reading https://github.com/Hypfer/Valetudo/issues/337#issuecomment-555672615 and https://github.com/Hypfer/Valetudo/issues/337#issuecomment-560081105, I'd like to ask if it would be possible for Valetudo to:

* Add the option to publish retained discovery messages, so HA's MQTT Discovery will work across reboots
* Add a `unique_id` so HA's entity setup can persist itself.

It's possible only one of the two options are needed, but "why not both" :)

Hypfer, 2020-03-14:
Discovery messages [are already retained](https://github.com/Hypfer/Valetudo/blob/master/lib/MqttClient.js#L168).

Adding the `unique_id` would definitely make sense. Question is: What should we use for it.
The official DID (device ID) would be a good candidate. As an alternative, we could probably also use the Wi-Fi MAC.

LordMike, 2020-03-14:
So odd. I've had issues that could be explained by not being retained .. HA doesn't know of my vacuum on restarts. Maybe RabbitMQ has some oddities here.. will check.

As for unique_id - both those options are good. Didn't consider those. Either is unique, buut, semantically the device id fits the bill better, right?... Maybe we'll have an ethernet-linked vacuum someday..

LordMike, 2020-03-14:
I've confirmed that RabbitMQ does not retain MQTT messages in the same way a real MQTT broker would. The queues RabbitMQ creates for you needs to be durable / persisted, else the retained message does not arrive to the subscriber. 

As HA doesn't do what's necessary for RabbitMQ to create durable queues, I've switched to mosquitto .. so there's that. Retaining works now.

The `unique_id` is needed for HA to persist its knowledge of the entity:

[image]

Hypfer, 2020-04-12:
This is in fact already included in master since e6fa41f

LordMike, 2020-04-12:
I got sidetracked from this. I'm not sure, now, why I saw what I saw. But it works now - so maybe it was old data. In any case. Great work - turns out all my qualms were already handled :).

closed 2020-04-12 by LordMike