Hypfer/Valetudo #461: Zones editor empty on opening 2nd zone

uowiec, 2020-03-20:
When trying to create or edit multiple zones in zone editor, I'm only able to edit 1 zone, after navigating to another zone I get empty map

## To Reproduce

1. Install Valetudo from commit 5dc6e438afc1361b1955230a854a8c677da01e45 
2. service valetudo stop, delete config.json, service valetudo start
3. open valetudo web page, wait for "no map data" map to be replaced with the actual map
4. go to zones tab
5. create a zone 'zone1', edit, mark a rectangle, save, navigate back
6. create another zone 'zone2', click edit - map disappears, and it's not possible to draw rectangles anymore (screenshot)
7. also when navigating back to map tab, the map disappears

### Screenshots
Entering zone editor for the 2nd zone
[image]

### Vacuum Model
Roborock S55

### Valetudo Version
0.4.0

### User-Agent
Same behaviour on both Chrome and Firefox
Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/73.0.3683.103 Safari/537.36
Mozilla/5.0 (X11; Ubuntu; Linux x86_64; rv:74.0) Gecko/20100101 Firefox/74.0

## Expected behavior
1. after creating zone1, editing it and saving, the zone is saved and one can enter it again and map and zone rectangle is shown
2. once another zone editor is entered, map should be shown and user should be able to edit and save the zone

## Additional context

A workaround is to restart valetudo service, then zone2 editor is available, after that entering zone1 editor again, causes map to disappear.

rumpeltux, 2020-03-21:
This will be fixed by ac9d49b8 (part of #457).

rumpeltux, 2020-03-21:
FTR: Reason is that the js is added to the page, but the top-level code apparently only executed once.
So consecutive loads don’t rerun the code.

leonardpitzu, 2020-03-22:
issue persists (also left comment in the pull req) - zone editor is empty also with the last commit

error point to forbidden-markers-configuration-map.js:2
`SyntaxError: import declarations may only appear at top level of a module`

rumpeltux, 2020-03-22:
Thanks. It’s a similar issue but for another page. I’ll prepage a fix.

commit 9238fe70 (2020-03-22): Fix marker configuration js loading (fixes #461)
closed 2020-03-22 by rumpeltux