Hypfer/Valetudo #484: Disabling sending map data over MQTT not working

rvdblink, 2020-04-15:
## Describe the bug
"provideMapData": false still sends data for the map over mqtt

## To Reproduce
Set up mqtt as below and check message going over mqtt for the map data

`"mqtt": {
    "enabled": true,
    "identifier": "rockrobo",
    "topicPrefix": "valetudo",
    "autoconfPrefix": "homeassistant",
    "broker_url": "mqtt://xxxxxxxx",
    "provideMapData": false,
    "caPath": ""
  },
`

### Screenshots

N/A
### Vacuum Model

S5
### Valetudo Version

0.4.0
### User-Agent

N/A
## Expected behavior

No map data over mqtt being send
## Additional context

Overload of mqtt data is crashing my raspberry pi 3b+ and is pinpointed to the map from valetudo.

Hypfer, 2020-04-15:
But how? I don't get it.

https://github.com/Hypfer/Valetudo/blob/master/lib/MqttClient.js#L141
The option is clearly getting checked here.

Hypfer, 2020-04-15:
@rvdblink
Could it be possible that you've used mqtt before disabling the map publishing?
If so, could it be a retained message which is causing these issues?

rvdblink, 2020-04-15:
> @rvdblink
> Could it be possible that you've used mqtt before disabling the map publishing?
> If so, could it be a retained message which is causing these issues?

Yes I have used it before, but than I wouldn't expect the counter of received messages to go up still? 
I have now removed the retained messages from MQTT explorer to see what it does when I start the robot in an hour or so...

Hypfer, 2020-04-15:
There's also another possibility

This is where the option is set
`this.provideMapData = mqttConfig.provideMapData !== undefined ? mqttConfig.provideMapData : true;`

And this is where it is checked
This statement only checks if provideMapData is a "truthy" value. The string "false" is a truthy value
`if (this.provideMapData) {`

```
> if("false"){true} else {false}
true
```

So if anyone would specify  `"provideMapData": "false"` it would actually be interpreted as true.

rvdblink, 2020-04-15:
Nope, that's not it either. I use "provideMapData": false

below you can see the mqtt message that was just received by mosquitto (cut off to not flood this thread):

`{"header_length":20,"data_length":55800,"version":{"major":1,"minor":0},"map_index":699,"map_sequence":12706,"image":{"position":{"top":145,"left":319},"dimensions":{"height":563,"width":448},"pixels":{"floor":[[323,502],[322,501],[323,501],[321,500],[322,500],[323,500],[324,500],[319,499],[320,499],[321,499],[322,499],[323,499],[324,499],[318,498],[319,498],[320,498],[321,498],[322,498]'

Hypfer, 2020-04-15:
You're completely right.

This was actually fixed on the same day 0.4.0 was released (a140eea)
There just wasn't any release since then which is why this is still showing up even though it's fixed in master since 9 months

rvdblink, 2020-04-15:
Oh duh... I’ll patch that myself tomorrow. Here down under its time to switch off :)

rvdblink, 2020-04-16:
And that works now... Thanks!

For future reference (and for others): https://github.com/Hypfer/Valetudo/commit/a140eeaa3956dec7cf548e3d1938e2ccb43a1e91 has this fixed but has not officially released in a release. Quick fix is to clone the repository, build the valetudo and copy it over to the robot.

closed 2020-04-16 by rvdblink