Hypfer/Valetudo #488: Possible solution for random firmware resets?

momo42, 2020-04-25:
I just wanted to point you to this possible solution for the firmware resets we are all facing from time to time: [https://www.roboter-forum.com/index.php?thread/33043-mein-roborock-s50-hat-sich-gestern-selbst-zur%C3%BCckgeflasht/&postID=565266#post565266](url)

User _niete_ posted in this (German) thread that he found in another forum (no idea where ...) that `fsck` is triggering the resets and that these changes should help:
```
# nano /etc/crontab

0 23    * * *   root    service rrwatchdoge stop
10 23   * * *   root    rm /mnt/data/rockrobo/rrlog/*REL -R
15 23   * * *   root    /root/check_all
0 4     * * *   root    service rrwatchdoge start

# nano /root/check_all

umount /mnt/updbuf
umount /mnt/data
umount /mnt/reserve
umount /mnt/default
fsck -y /dev/mmcblk0p10
fsck -y /dev/mmcblk0p1
fsck -y /dev/mmcblk0p11
fsck -y /dev/mmcblk0p6
mount /dev/mmcblk0p10 /mnt/updbuf
mount /dev/mmcblk0p1 /mnt/data
mount /dev/mmcblk0p11 /mnt/reserve
mount -o ro /dev/mmcblk0p6 /mnt/default

# service cron restart
```

Please excuse that I didn't put this into one of the many reset issues. Didn't know where the best place is so that it doesn't get lost. My parents' v1 is also affected but I'm not able to test if this solution really works due to COVID-19 social distancing. Can anyone of you test this and give his thoughts on this?

dpalecek, 2020-04-25:
Just had a firmware reset and reflashing it now. I will try this and report back.

mlq, 2020-04-29:
The script seems to come from here: https://github.com/MadJoker0815/roborock_nologs - I did some more reading and for some people it apparently helped for multiple weeks but it did not seem to fix the issue once and for all.

michaelotto, 2020-04-29:
Well, what this script does is

a) check and fix some (not all!) of the mounted file systems and repair them if broken and possible and
b) prevent the nightly reboots.

Without the nightly reboots, no fsck is done at night and thus the resets cannot happen any more. But the suppression of the nightly reboots may have unintended side effects, as we don't know the real purpose of the nightly reboots as intended by Xiaomi. It's probably just to have a clean and determined state every day and prevent cascading errors due to sloppy design and programming, which may happen if you don't reboot.

Anyway, I have these changes in effect and have never had a firmware reset. I reboot the bot manually about once a week and keep my fingers crossed it'll comes back.

It is not a real solution to the problem though. The original firmware survives the nightly reboots without resets, Valetudo - in some case - does something differently and obviously breaks the file system. So this fix does not cure the root problem.

Hypfer, 2020-05-05:
Superseded by further discussion in #206

closed 2020-05-05 by Hypfer