Hypfer/Valetudo #497: Properly support running valetudo on a different host

Hypfer, 2020-04-30:
While this is already possible for development purposes, having a clean and supported way to do this would be preferable.

### Motivation 
It's getting harder to modify robot firmwares. There might even be new robots which don't fulfill the hardware requirements or maybe even run a custom RTOS instead of linux.

Also, factory resets wouldn't be an issue anymore for roborock devices.

### Goals

* Everything required (depending on vacuum model) should be configurable via the configuration file.
  Different robots will have different requirements so this must be considered.
  Roborock vacuums for example require these:
  * Model
  * CloudKey
  * DID
* Valetudo shouldn't care about where it is running. There must be no assumptions about the host
* There should be a dockerfile for easy setup
* Documentation has to be provided
  * How to set this up (obv.)
  * How to get the required secrets (e.g. CloudKey)

### Misc

Redirecting traffic to the valetudo instance is out-of-scope for the project. This shall be done by whatever the user might use.
A fairly smart router should be able to redirect the required traffic. If not, having something like a rpi with a custom AP + iptables just for the robot would also be a possibility.

Hypfer, 2020-04-30:
Stuff that has to run on the vacuum could be moved to a seperate service which provides a simple rest API.

e.g. Swapping Maps

Hypfer, 2020-06-02:
The local token is published to the cloud interface once on the initial connection.

The method is `_otc.info` and the whole message looks like this:

```json
{
	"id": 123,
	"params": {
		"ap": {
			"ssid": "myWifi",
			"bssid": "F8:1A:67:CC:BB:AA",
			"rssi": -30
		},
		"hw_ver": "Linux",
		"model": "rockrobo.vacuum.v1",
		"netif": {
			"localIp": "192.168.1.205",
			"gw": "192.168.1.1",
			"mask": "255.255.255.0"
		},
		"fw_ver": "3.3.9_003077",
		"mac": "34:CE:00:AA:BB:DD",
		"token": "xxx"
	},
	"method": "_otc.info"
}
```

See https://dgiese.scripts.mit.edu/talks/BeVX-2018/BeVX_prepublic.pdf for reference.

This token does not change unless the robot is provisioned again. Provisioning includes changing the wifi credentials.

Hypfer, 2020-06-02:
Since we're the cloud now, the whole tokenProvider token refresh process is obsolete.

It only exists so that valetudo could be used to do the initial wifi provisioning anyways, which would require a new solution for this approach

andreashaertel, 2020-06-26:
I am new to the topic, so maybe my question is stupid: 
The idea is to install Valetudo on any host machine, set up the configuration for a specific vacuum cleaner model and vendor, then use for instance a FritzBox WiFi and redirect all traffic of the vacuum cleaner to the host on which Valetudo is installed? 

Would this work (or does it already)? 
I have a S5 Max and yet no idea how exactly it communicates with the App and the Chinese sever. 
Thank you for any answer

andreashaertel, 2020-06-28:
Well, I found the  OpenMiHome / mihome-binary-protocol on github and I tried to speak to my robot: 
1. I have connected my Laptop via WiFi with my Roborock S5 Max.
2. Then I have send a UDP "Hello" package to the IP 192.168.8.1 (from this address I received a positive ping) and port 54321 and listened with a second programm on the port 54321 on my local machine - receiving nothing. According to the OpenMiHome project I would have expected a reply package from the robot. Does anyone has experience with such kind of communication with the robot? Thank you

Hypfer, 2020-06-28:
I'm sorry but this really isn't the place to thoroughly talk about the miio protocol and the inner workings of the robot.

There's some information in the [docs](https://valetudo.cloud) and there's of course [dennis' talk](https://media.ccc.de/v/34c3-9147-unleash_your_smart-home_devices_vacuum_cleaning_robot_hacking)

Anyways BTT.

To use an S5Max (or any roborock vacuum) with Valetudo running on a different host (which is the only option for the S5Max atm), you need to dump its cloudKey + DID which are the shared secrets which were provisioned in the factory.
This requires a lot of disassembly.

When you've obtained this data, you also need a router/network/etc, capable of doing the required traffic redirection and dns changes.
Then, you can simply run Valetudo on whatever you like by simply configuring the cloudKey and DID in the config file and redirecting the traffic to it. This is already possible

There will probably be a guide on how to do that at some point but since this ticket is still open, this hasn't happend yet.

rumpeltux, 2020-06-29:
> When you've obtained this data, you also need a router/network/etc, capable of doing the required traffic redirection and dns changes.
> Then, you can simply run Valetudo on whatever you like by [...]

You can do all of that on the robot itself, without requiring to mess with your network setup.
There’s instructions for Viomi: http://valetudo.cloud/pages/installation/viomi.html#robot-setup

For Roborock see https://github.com/Hypfer/Valetudo/tree/master/deployment/etc for the scripts used.
Just need to change the destination IP there (and in your local valetudo config).

Hypfer, 2020-06-29:
@rumpeltux Yes but not on the S5 Max atm

andreashaertel, 2020-06-29:
Thank you very much for the link to Dennis' CCC video. 
I hope my discussion is not off topic but on the way to the goal! 

For now, I do not see a chance to get access to the cloud key without opening the robot - too bad. 
I guess there is no idea to extract it at the moment? 
Would it work to set up a network without internet access and work with a mobile phone (xiomi app installed, just WiFi) to command the robot? Alternatively, sending self-made commands via udp should work as well, right? 

However, the first handshake should also have worked between Laptop and robot via udp (as explained above) - so I wonder what went wrong here. I opened an issue here ( https://github.com/OpenMiHome/mihome-binary-protocol/issues/9 ) but still waiting for reply. 

It would be a first step to be able to command the robot within an encapsulated network - or does the robot require the cloud?

Hypfer, 2020-06-30:
Please refer to the telegram group, a roboter-forum thread etc. since this is way out of scope for this issue.

I'd strongly encourage you to read the introduction section of [the legacy installation guide](https://valetudo.cloud/pages/installation/installation-instructions.html) to shine some light on this.

The local interface is known and implementations to interface with it do exist as long as you have the token.
* https://forum.fhem.de/index.php/topic,73052.0.html
* https://www.home-assistant.io/integrations/vacuum.xiaomi_miio/
* https://github.com/rytilahti/python-miio

just to name a few

andreashaertel, 2020-07-01:
Great: extracting the token works and thus I should be able to control the robot. 

So: why do I need the connection to the cloud (and the CloudKey)? I can simply block all traffic to the Cloud and just use the local interface. 
Of course, I do not get the map by this: but I can ask the robot to send me its status via the local interface and this should contain the current coordinates. Thus, I could scan this coordinates over some cleaning processes and draw my own map. In accordance to this map, I can define zones and communicate them to the robot. I do not see any problems in the point that the robot has a slightly different map than I would have, because in the end I just need the coordinates in a map of my appartment. 

If this would work, we do not need the Cloud communication and, accordingly, we do not need to change the robots firmware and software. We just would have to block the communication between the robot and the Cloud. It could even work to filter the network traffic in such a way that firmware updates would be possible without allowing any data storage to the cloud. 

Then valetudo could always run just on any host in the network. 
Now, as you can read above, I am quite new to the topic and you are more experienced: would this work?

rumpeltux, 2020-07-01:
There are some parts that don’t work on the local interface. I know at least of

* maps
* status push notifications (vs `get_status` poll)
* firmware updates

It’s not unlikely that manufactures will restrict the local interface further over time if that’s not already happening, so I don’t think it’s a viable long-term approach. That said, it sounds workable in your scenario but I’m not sure if that’s a viable approach for everyone (also all code to track robot movement to infer the map would still need to be written).

Hypfer, 2020-07-01:
Trying to guess the internal robot state is at best imprecise and at worst absolute nonsense.

Also, this is off-topic ffs.

closed 2021-01-09 by Hypfer