Hypfer/Valetudo #499: Power mode is undefined after 0.5.0 update

definitio, 2020-04-30:
## Describe the bug

## To Reproduce

### Screenshots
[image]

### Vacuum Model
Gen1

### Valetudo Version
0.5.0

Hypfer, 2020-04-30:
This is probably caused by the slightly different presets on Gen 1.

If you try setting it to your prefered speed again, this should vanish

definitio, 2020-04-30:
I have tried. It doesn't work.

pkishino, 2020-05-02:
seeing same issue here, gen1

Hypfer, 2020-05-02:
What base firmware are you guys on?

I should probably add that to the bug issue template

pkishino, 2020-05-02:
FW version 3.5.4
FW build 004007

Hypfer, 2020-05-02:
The raw response to the `get_status` miio command would be very helpful.

There are multiple ways to get that. [python-miio](https://python-miio.readthedocs.io/en/latest/vacuum.html#raw-commands) is one of them

pkishino, 2020-05-02:
Sending cmd get_status with params []
[{'msg_ver': 8, 'msg_seq': 477, 'state': 8, 'battery': 100, 'clean_time':
0, 'clean_area': 0, 'error_code': 0, 'map_present': 1, 'in_cleaning': 0,
'fan_power': 75, 'dnd_enabled': 1}]

Hypfer, 2020-05-02:
`'msg_ver': 8`

`if (msg_ver >= 3) {`
https://github.com/Hypfer/Valetudo/blob/04e1720f3c5e4d1061c61781451d9dcd0bfd6fd3/lib/devices/Roborock.js#L375

Okay that's the issue here. But how many msg_ver are there? And why does Gen1 has such a high number

closed 2020-05-02 by Hypfer

pkishino, 2020-05-02:
Can this be be manually updated on the robot? Couldn’t find the file :/

pkishino, 2020-05-03:
For me it seems to have fixed itself..I see the normal power modes now but it’s a bit slow to switch from HA. But it works :)

PR #502 (2020-05-02): fix: gen1 fanspeeds
should fix #499