Hypfer/Valetudo #500: Stuck in "Loading state!"

wayller, 2020-04-30:
## Describe the bug

### Screenshots

### Vacuum Model

S50

### Valetudo Version

0.5.0

## Log
[2020-04-30T11:54:26.546Z] [INFO] exiting...
[2020-04-30T11:54:26.750Z] [INFO] exiting...
[2020-04-30T11:54:27.783Z] [INFO] Loading configuration file: /mnt/data/valetudo/config.json
[2020-04-30T11:54:27.824Z] [INFO] Starting Valetudo for Vacuum Model roborock.vacuum.s5
[2020-04-30T11:54:27.826Z] [INFO] DID 
[2020-04-30T11:54:27.827Z] [INFO] CloudKey 
[2020-04-30T11:54:28.132Z] [INFO] exiting...
[2020-04-30T11:54:28.210Z] [INFO] Loading configuration file: /mnt/data/valetudo/config.json
[2020-04-30T11:54:28.245Z] [INFO] Starting Valetudo for Vacuum Model roborock.vacuum.s5
[2020-04-30T11:54:28.247Z] [INFO] DID 
[2020-04-30T11:54:28.247Z] [INFO] CloudKey 
[2020-04-30T11:54:28.477Z] [INFO] exiting...
[2020-04-30T11:54:28.736Z] [INFO] Loading configuration file: /mnt/data/valetudo/config.json
[2020-04-30T11:54:28.769Z] [INFO] Starting Valetudo for Vacuum Model roborock.vacuum.s5
[2020-04-30T11:54:28.770Z] [INFO] DID 
[2020-04-30T11:54:28.770Z] [INFO] CloudKey 
[2020-04-30T11:54:28.966Z] [INFO] exiting...
[2020-04-30T11:54:35.146Z] [INFO] Loading configuration file: /mnt/data/valetudo/config.json
[2020-04-30T11:54:35.201Z] [INFO] Starting Valetudo for Vacuum Model roborock.vacuum.s5
[2020-04-30T11:54:35.203Z] [INFO] DID 
[2020-04-30T11:54:35.204Z] [INFO] CloudKey 
[2020-04-30T11:54:35.301Z] [INFO] Loading configuration file: /mnt/data/valetudo/config.json
[2020-04-30T11:54:35.339Z] [INFO] Starting Valetudo for Vacuum Model roborock.vacuum.s5
[2020-04-30T11:54:35.341Z] [INFO] DID 
[2020-04-30T11:54:35.341Z] [INFO] CloudKey

wayller, 2020-04-30:
[2020-04-30T12:07:35.279Z] [INFO] Loading configuration file: /mnt/data/valetudo/config.json
[2020-04-30T12:07:35.306Z] [INFO] Starting Valetudo for Vacuum Model roborock.vacuum.s5
[2020-04-30T12:07:35.307Z] [INFO] DID #########
[2020-04-30T12:07:35.307Z] [INFO] CloudKey #########
[2020-04-30T12:07:35.493Z] [INFO] Dummycloud is spoofing 203.0.113.1:8053 on 127.0.0.1:8053
[2020-04-30T12:07:35.499Z] [INFO] Webserver running on port 80

Got this in the log now.

Hypfer, 2020-05-01:
So does it work now?

If not, try setting the `logLevel` to `debug` in the config and check if there's more in the logs

wayller, 2020-05-01:
No, it doesn't work.

[2020-05-01T14:00:31.683Z] [INFO] CloudKey ###
[2020-05-01T14:00:31.721Z] [DEBUG] >>> local: HandshakePacket()
[2020-05-01T14:00:31.857Z] [INFO] mqtt start
[2020-05-01T14:00:32.083Z] [INFO] exiting...
[2020-05-01T14:00:39.645Z] [INFO] Loading configuration file: /mnt/data/valetudo/config.json
[2020-05-01T14:00:39.671Z] [INFO] Starting Valetudo for Vacuum Model roborock.vacuum.s5
[2020-05-01T14:00:39.672Z] [INFO] DID ####
[2020-05-01T14:00:39.672Z] [INFO] CloudKey ####
[2020-05-01T14:00:39.709Z] [DEBUG] >>> local: HandshakePacket()
[2020-05-01T14:00:39.844Z] [INFO] mqtt start
[2020-05-01T14:00:40.065Z] [INFO] exiting...
[2020-05-01T14:00:39.709Z] [DEBUG] >>> local: HandshakePacket()
[2020-05-01T14:00:47.643Z] [INFO] Loading configuration file: /mnt/data/valetudo/config.json
[2020-05-01T14:00:47.669Z] [INFO] Starting Valetudo for Vacuum Model roborock.vacuum.s5
[2020-05-01T14:00:47.670Z] [INFO] DID ###
[2020-05-01T14:00:47.670Z] [INFO] CloudKey ####
[2020-05-01T14:00:47.708Z] [DEBUG] >>> local: HandshakePacket()
[2020-05-01T14:00:47.846Z] [INFO] mqtt start
[2020-05-01T14:00:48.074Z] [INFO] exiting...

Hypfer, 2020-05-01:
Why is it terminating after one second?

Since it's missing the `Valetudo shutdown in progress...` message, it's not because of a `SIGTERM` nor a `SIGINT`.

Maybe it couldn't bind to a port? 

Check `netstat -tulpen`

wayller, 2020-05-01:
```
root@rockrobo:~# netstat -tulpen
Active Internet connections (only servers)
Proto Recv-Q Send-Q Local Address           Foreign Address         State       PID/Program name
tcp        0      0 127.0.0.1:5037          0.0.0.0:*               LISTEN      1357/adbd
tcp        0      0 127.0.0.1:54322         0.0.0.0:*               LISTEN      1555/miio_client
tcp        0      0 127.0.0.1:54323         0.0.0.0:*               LISTEN      1555/miio_client
tcp        0      0 0.0.0.0:22              0.0.0.0:*               LISTEN      1358/dropbear
tcp        0      0 127.0.0.1:55551         0.0.0.0:*               LISTEN      1564/rriot_tuya
tcp        0      0 0.0.0.0:55553           0.0.0.0:*               LISTEN      1352/xdnsmasq
tcp        0      0 :::80                   :::*                    LISTEN      1954/valetudo
tcp        0      0 :::22                   :::*                    LISTEN      1358/dropbear
tcp        0      0 :::55553                :::*                    LISTEN      1352/xdnsmasq
udp        0      0 0.0.0.0:55553           0.0.0.0:*                           1352/xdnsmasq
udp        0      0 0.0.0.0:55559           0.0.0.0:*                           1564/rriot_tuya
udp        0      0 0.0.0.0:6669            0.0.0.0:*                           1564/rriot_tuya
udp        0      0 0.0.0.0:54321           0.0.0.0:*                           1555/miio_client
udp        0      0 127.0.0.1:8053          0.0.0.0:*                           1954/valetudo
udp        0      0 0.0.0.0:60567           0.0.0.0:*                           1954/valetudo
udp        0      0 :::55553                :::*                                1352/xdnsmasq

```

Hypfer, 2020-05-01:
Hm since it runs now that wasn't informative.

I assume that it's something wrong with the built firmware

wayller, 2020-05-01:
I'll reflash it, that's faster than finding out what is wrong.

Hypfer, 2020-05-03:
Does it work now?

wayller, 2020-05-03:
Reflashed last night, and it worked. Now it's back to loading state. Didn't do anything.

wayller, 2020-05-09:
Flashed a prebuilt image a few days ago. No problems so far.

closed 2020-05-09 by wayller