Hypfer/Valetudo #501: Support for adjustable water output of newer roborock vacuums

Hypfer, 2020-05-02:
Newer models (not yet rootable) have additional status properties as well as controls for the mop function.

Quoting https://github.com/openhab/openhab-addons/issues/6628

`get_status` response
```
{
            "msg_ver": 2,
            "msg_seq": 1433,
            "state": 2,
            "battery": 93,
            "clean_time": 40,
            "clean_area": 1127500,
            "error_code": 0,
            "map_present": 1,
            "in_cleaning": 1,
            "in_returning": 0,
            "in_fresh_state": 0,
            "lab_status": 1,
            "water_box_status": 0,
            "fan_power": 101,
            "dnd_enabled": 0,
            "map_status": 3,
            "is_locating": 0,
            "lock_status": 0,
            "water_box_mode": 200,
            "water_box_carriage_status": 0,
            "mop_forbidden_enable": 0
 }
```
The new ones are:

* `water_box_status`
* `lock_status`
* `water_box_mode`
* `water_box_carriage_status`
* `mop_forbidden_enable`

Both `water_box_status` as well as `lock_status`  are also availabe on the regular Gen2 S5.

It might make sense to log an info if there are additional and unhandled properties in any robot reply to detect these changes faster.

To further quote the linked openhab issue:

`water_box_mode` can take the following values:
* 200 - Off
* 201 - Low
* 202 - Medium
* 203 - High
* 204 - Customize (Auto)

The customize switch is only switched on if both fan_power and water_box_mode are set to the corresponding values (fan=106 & water=204).

[image]

thomas725, 2021-03-02:
I tried switching water_box_mode from 202 (which seems to be the default for my S5 max) to 203 by editing /mnt/data/rockrobo/RoboController.cfg - but it is always overwritten back to 202. Is there a different place where I can set this in a way that it persists?

thomas725, 2021-03-02:
Ah, got a solution in the Telegram Group:
`miiocli vacuum --ip ******* --token ****** set_waterflow maximum`

Hypfer, 2021-03-19:
fdebd3c4

closed 2021-03-19 by Hypfer