Hypfer/Valetudo #505: No build and deploy instructions

schlamar, 2020-05-05:
I think it's essential for an open source product to have clear instructions how to build it on your own. Right now, the installation instructions just say that you should download the binary and offer no further details on how to build the binary. 

I guess there are a lot of programmers with client side JS experience but never worked with npm / server side JS. So they have no clue how to get started and this might turn away possible contributors.

Looking at the travis configuration I assume it is something simple like

- npm install
- npm run build
- replace /usr/local/bin/valetudo
- service valetudo restart

but it would be great if these steps are in an official documentation.

And what are the dependencies, do you need a specific node.js version?

Additionally, some debugging tipps for new contributors (log configuration etc.) might help getting new contributors .

Hypfer, 2020-05-05:
https://github.com/Hypfer/Valetudo/blob/master/develop/HOWTO.md

🤯

schlamar, 2020-05-05:
Ah yes, exactly what I'm looking for. 👍 

However, AFAIS this isn't linked anywhere (Readme, Homepage, ...)?!

Hypfer, 2020-05-05:
It's also outdated since there's no dockerfile anymore. Definitely room for improvement there.

If you feel like it, it would be great if you could move that over to the docs section [here](https://github.com/Hypfer/Valetudo/tree/master/docs) so that it can be found in [the docs](https://hypfer.github.io/Valetudo/)

Hypfer, 2020-05-16:
https://hypfer.github.io/Valetudo/pages/development/building-and-modifying-valetudo.html

commit 3cdfbbb2 (2020-05-16): docs: Updated documentation to fix #505 and #512
closed 2020-05-16 by Hypfer