Hypfer/Valetudo #509: MQTT password visible in WebUI

xoxys, 2020-05-06:
## Describe the bug
After upgrading to the latest valetudo release we are able to configure mqtt from the UI. While this is pretty cool in general, I'm not really sure if it is a good idea to show the connection credentials/password in clear text to everyone who could access the web frontend... Would be good to have the ability to disable web editing of the mqtt part completely and/or split out the connection URL into different fields without displaying the current password (as it is done for the wifi config site).

### Valetudo Version
0.5.1

rumpeltux, 2020-05-08:
@TheLastProject FYI in case you want to tackle this too.

TheLastProject, 2020-05-08:
I mean, to be fair, anyone who can access the vacuum can just call `/api/get_config` which will also show the password in plain text, so it's not like this information wasn't exposed before, but it couldn't hurt to split it up yeah.

To do it properly though, in a way that would make it hideable in the API too, the broker_url config variable would need to be split into multiple values: https://github.com/Hypfer/Valetudo/blob/master/lib/MqttClient.js#L113

Or it should be possible to password-protect the API too?

I'm not opposed to fixing this, but I feel fixing this only in the web UI would give a false sense of security as this is still exposed over the API.

xoxys, 2020-05-08:
Good point... Was not aware of the API endpoint.

PR #520 (2020-05-15): feat(mqtt): split broker_url in multiple fields
and hide passwords

resolves #509 
resolves #389
closed 2020-05-15 by Hypfer