Hypfer/Valetudo #510: /etc/rc.local is broken for 2008+ firmwares

Hypfer, 2020-05-06:
Execution stops right after `ip6tables        -F OUTPUT` 

`/etc/rc.local: line 22: ip6tables: not found`

This breaks the dummycloud interface and therefore the map view

Hypfer, 2020-05-06:
Simply rearranging the commands doesn't help either

```
[root@rockrobo ~]# /etc/rc.local
iptables v1.4.21: unknown option "--to-destination"
```

pidator, 2020-05-06:
> `/etc/rc.local: line 22: ip6tables: not found`

if my memory serves me right @zvldz provides the ip6tables function in his 2008 images by including [addon2.tgz](https://github.com/zvldz/vacuum/blob/master/custom-script/files/addon2.tgz).

Hypfer, 2020-05-06:
@pidator 

You're correct. And it's not only that. The whole DNAT functionality is missing in 2008+ firmwares. The modules to restore that are also part of that tar file.

Roborock actually removed that with the intention of preventing the usage of Valetudo. 😶

pidator, 2020-05-06:
> Roborock actually removed that with the intention of preventing the usage of Valetudo.

There are definitely some more firmware modifications necessary as before to get everything working again. Check also [this part](https://github.com/zvldz/vacuum/blob/master/custom-script/custom_dns_catcher.sh)!

Probably a reason for the providing of prebuild image files...

rumpeltux, 2020-05-08:
Is this problem solved by #508 or is that fix unrelated?

Hypfer, 2020-05-08:
Unrelated. This was fixed by modifying dustbuilder to also apply parts of https://github.com/zvldz/vacuum/blob/master/custom-script/files/addon2.tgz to restore the functionality.

It's not a valetudo but rather a firmware issue. Thanks roborock

closed 2020-05-08 by Hypfer