Hypfer/Valetudo #512: How can I update Valetudo

Wolf6660, 2020-05-08:
how can I update form Valetudo 0.4 to 0.5.1?

thanks

Daniel

Hypfer, 2020-05-08:
[image]
[image]

Why won't anyone read that. I just don't understand

rumpeltux, 2020-05-08:
Rephrased it as a bug ;-)

bmansvk, 2020-05-14:
I have the same question. There is no info how to upgrade to the new version of Valetudo. I think that building vacuum image from the scratch is not required, because of already working SSH access. 

Is is sufficient to replace "valetudo" binary from the new release in a vacuum?

alexkn, 2020-05-14:
Yes, it is sufficient to replace the binary, as long as you not want to update the firmware anyway.

Just stop valetudo before.
Depending on firmware:
`/etc/init/S11valetudo stop`
or
`service valetudo stop`

Hypfer, 2020-05-16:
Depending on your setup, it may or may not be necessary to reflash completely.

Since there might be new changes/fixes in the image builders, it is however recommended to so

Hypfer, 2020-05-16:
https://hypfer.github.io/Valetudo/pages/general/upgrading.html

closed 2020-05-16 by Hypfer