Hypfer/Valetudo #521: Configure Iterations for Zoned Cleanup

Hypfer, 2020-05-15:
Both saved zones as well as dynamically drawn ones are currently missing the option to configure the Iteration count.

Hypfer, 2021-01-09:
This only needs UI Support. The backend already supports this parameter

IronicBadger, 2021-07-23:
If the backend already supports this, presumably I could call this via `mqtt.publish` like so:

```
service: mqtt.publish
data:
  topic: valetudo/robovacDown/MapSegmentationCapability/clean/set
  payload: '["17", "20"]'
```

What would the payload need to be to clean zone 17 3 times say?

Hypfer, 2021-07-29:
@IronicBadger It's not yet possible via mqtt to specify iterations or other parameters

closed 2021-10-11 by Hypfer

bllngr, 2022-01-09:
Has this issue been closed because it has been implemented in the meantime? I didn't find anything in the UI. @Hypfer's [comment above](https://github.com/Hypfer/Valetudo/issues/521#issuecomment-757133082) mentions support in the backend, could it be set (permanently) in some configuration file?

Hypfer, 2022-01-09:
@bllngr 

[image]

Yes. It needs firmware support though