Hypfer/Valetudo #527: Not working "Close" and "Cancel" buttons on the "Timers" page

gvolt, 2020-05-29:
## Describe the bug
On the "Timers" page the "Close" and "Cancel" buttons in the dialogs used to set **timezone**, add **cleaning timers** and edit the **do not disturb timer** do not work (do not close the dialogs)

## To Reproduce
Steps to reproduce the behavior: - Example for timezone setting -
1. Go to "Timers"
2. Click on the cog symbol in the right corner
3. Click on the close button below the timezone selector
4. Nothing happens (dialog does not close) and the "Timers" page can only be left using the back button of the browser

### Screenshots

### Vacuum Model
Gen2/S5

### Valetudo Version
0.5.2

### Firmware Version
2008

### User-Agent
Firefox 76.0.1; Windows 10 Professional; x64
Firefox 76.0.1; openSUSE Tumbleweed with Gnome and KDE Plasma (updated 2020.05.28); x64
Fennec 68.8.0 F-Droid; Android 10 (AOSP)

## Expected behavior
The timezone selection dialog should close.

## Additional context
The same issue exists for the "Cancel" buttons of the dialogs for adding "cleaning timers" and for editing the "do not disturb timer". In both cases the "Cancel" buttons do not close the dialogs and one needs to use the browser's back button to leave the "Timers" page.

Hypfer, 2020-05-29:
Fixed. Thank you for reporting

closed 2020-05-29 by Hypfer

gvolt, 2020-05-29:
That was fast :D Thanks for fixing. Keep up the great work. 👍

commit e2ea1ce2 (2020-05-29): fix(ui): Fix #527 not working close and cancel buttons in timer settings