Hypfer/Valetudo #528: Documentation issues in the firmware installation guide

bbarbisch, 2020-05-29:
This is not about a bug in valetudo but about some Issues with the documentation of the firmware update I ran into during todays upgrade session of my gen2 (upgrade to valetudo 0.5.2):

*  Recommended Firmware Version
  * The legacy page recommends firmware version 1886 for gen2 vacuums but nothing is mentioned on the main installation guide
* duplicate --ip parameter in the mirobo upgrade-firmware command
* the builder_vacuum.sh call offers only a subset of the available parameters

I think the install guide should at least mention the various versions available and their compatibility.  There seems to be a couple of versions out there and dustbuilder offers also v2008 and v2020 besides v1886 which may (mis)lead people to use maybe unsupported versions. And there is also [https://vacuumz.info/download/gen2/](https://vacuumz.info/download/gen2/) which offers prebuilt firmware images.

From what I've experienced at least v2008 has some major drawbacks:
* no ssh server installed (its a stripped-down version of  ubuntu 14.04. lts)
* no OTA firmware upgrade (mirobo ... upgrade-firmware fails, see also https://github.com/zvldz/vacuum/issues/14)
  * the builder script from zvldz offers a separate parameter --enable-local-ota to fix this (since 20200512. This applies a binary patch to files specific to v2008 (see https://github.com/zvldz/vacuum/issues/26)
* there are at least some issues with configuring timers on v2008 and probably more on other areas (didn't test further)

So to summarize (and for TLDR);
* Shall there be a firmware version recommendation in the installation guide?
  * if yes, which version(s)? (v1886, v1898, v2008,...)
* Shall the builder_vacuum command stay there as it is?
  * there is also the script template https://github.com/zvldz/vacuum/blob/master/vacuum_valetudo.sh which calls the builder script with some (all?) options but needs to be adjusted before execution (E.g. keyfiles)

If we come to an agreement which information shall be in the docs, I'm volunteering update them.

Hypfer, 2020-06-04:
Those are quite a few different things. Lets step through them one by one

>Recommended Firmware Version

The example URL currently contains the recommended firmware version but we can and should make that more clear.

>The legacy page recommends firmware version 1886 for gen2 vacuums but nothing is mentioned on the main installation guide

Thats because that file was just copy-pasted so that it didn't vanish when the wiki got disabled. it could probably be deleted

>duplicate --ip parameter in the mirobo upgrade-firmware command

That would be fixed by deleting the legacy guide

>the builder_vacuum.sh call offers only a subset of the available parameters

These are the recommended parameters.

>I think the install guide should at least mention the various versions available and their compatibility. 

If the firmware is listed in the guide, it is supported. That could also be added as a note.

>no ssh server installed 

Something went wrong in the build process then

> no OTA firmware upgrade

That's actually very good to know. I didn't know that. I'll talk to dennis about that.

> there are at least some issues with configuring timers on v2008 and probably more on other areas

It's just the timers that behave slightly different.  There's something new regarding the handling of the `*` character.

I'm still thinking about dropping native timer support altogether since there's really nothing to be gained by using the firmware feature but that's a topic for a different issue

alexkn, 2020-06-04:
> > duplicate --ip parameter in the mirobo upgrade-firmware command
> 
> That would be fixed by deleting the legacy guide

That is not a duplicate, the first --ip parameter is the address of the robot, the second is the address of the host you are using.

jansohn, 2020-06-17:
> It's just the timers that behave slightly different. There's something new regarding the handling of the `*` character.
>
> I'm still thinking about dropping native timer support altogether since [...]

Can you elaborate a little more on how one should use timers from firmware version 2008+? I'm not able to set timers after factory resetting and upgrading my robot (Gen2) to firmware v2008 and Valetudo 0.5.3.
Is there a way to simply write them into a configuration file on the robot?

Hypfer, 2020-06-18:
>Can you elaborate a little more on how one should use timers from firmware version 2008+?

It requires a fix in valetudo :/

nrj4all, 2020-07-14:
I would suggest to note a hint where the new firmware build is located after successful creation. (Step: Image building => ./builder_vacuum.sh)

It's just a "stupid user" issue but I wondered why I flashed my Robot with the original firmware multiple times cause I didn't knew that the valetudo'ed image is located in vacuum/output folder. I scratched my head about that and felt like a total retard after found out the truth.

closed 2021-01-09 by Hypfer