Hypfer/Valetudo #529: Missing mapping for state "going to target"

hillbicks, 2020-05-30:
Hey guys,

I'm currently in the process of automating my vacuum and stumbled upon a problem with the states of the vacuum. Looking at https://github.com/Hypfer/Valetudo/blob/master/lib/MqttClient.js for the state definitons, go_to_target currently maps to cleaning. It would be really great if at least go_to_target could be changed to have it's own state in HA. I searched for target in the issues here and in the telegram group, but couldn't find anything related to my issue. 

`const HA_STATES = {
    CLEANING: "cleaning",
    PAUSED: "paused",
    IDLE: "idle",
    RETURNING: "returning",
    DOCKED: "docked",
    ERROR: "error",
    ZONE_CLEANUP: "cleaning"
};

// Codes as per Status.js
const HA_STATE_MAPPINGS = {
    "CHARGER_DISCONNECTED": HA_STATES.IDLE,
    "IDLE": HA_STATES.IDLE,
    "CLEANING": HA_STATES.CLEANING,
    "MANUAL_MODE": HA_STATES.CLEANING,
    "SPOT_CLEANING": HA_STATES.CLEANING,
    "GOING_TO_TARGET": HA_STATES.CLEANING,
    "ZONED_CLEANING": HA_STATES.ZONE_CLEANUP,
    "RETURNING_HOME": HA_STATES.RETURNING,
    "DOCKING": HA_STATES.RETURNING,
    "CHARGING": HA_STATES.DOCKED,
    "CHARGING_PROBLEM": HA_STATES.ERROR,
    "ERROR": HA_STATES.ERROR,
    "PAUSED": HA_STATES.PAUSED,
};`

Hypfer, 2020-05-31:
Problem is that HA doesn't support such thing.

See: https://www.home-assistant.io/integrations/vacuum.mqtt/#state-mqtt-protocol

State has to be one of vacuum states supported by Home Assistant:

- cleaning,
- docked,
- paused,
- idle,
- returning,
- error.

hillbicks, 2020-05-31:
Why would they limit it that way? Doesn't make any sense to me. Thanks for the answer, I'll open a feature request on the ha-side and close this issue. In case this changes, I'll come back here :)

closed 2020-05-31 by hillbicks

Hypfer, 2020-06-01:
I guess it just wasn't something existing at the time where the component was created

hillbicks, 2020-06-02:
Thanks for your answer Hypfer, I opened up a new issue for HA.

https://github.com/home-assistant/core/issues/36373

Feel free to add a comment in case I missed something.

hillbicks, 2020-06-02:
Wrong place to post the feature request (enhancement). Opened up a new feature request on the HA forum

https://community.home-assistant.io/t/enhance-available-mqtt-vacuum-states/201448