Hypfer/Valetudo #530: Virtual walls are rotated after saving

Hacki1111, 2020-06-04:
The map and the zones are working fine on my vacuum with valetudo.
But I keep having issues with my carpets and virtual walls.
The robot should drive around the carpets so I have to set virtual walls around the carpets.
But some walls are rotated after saving. Where can be the problem?

Before I saved the setting:
[image]

After saving:
[image]

Hypfer, 2020-06-04:
If you're reporting a bug, please use the bug report template.

[image]
[image]
[image]

By answering the questions in that template, you're making this a lot easier for everyone.

This issue is fixed in 6724e555 which is included in `0.5.2`.

Since you haven't used the template, I have no idea which version you're running but it's most likely not `0.5.2`

Hacki1111, 2020-06-04:
I will try 0.5.2, still using 0.5.1.
And I will do it better next time. Thanks a lot!

closed 2020-06-04 by Hacki1111