Hypfer/Valetudo #532: Setting timezone does not work as expected

schlamar, 2020-06-06:
## Describe the bug

Setting timezone does not work.

## To Reproduce

Use dustcloud image which defaults to Asia/Singapore and set timezone in Timers to Europe/Berlin.

### Vacuum Model

Gen1

### Valetudo Version
0.5.2

### Firmware Version
4007

## Expected behavior

Timezone is set in all relevant linux configuration files.

## Actual behavior

Linux timezone config /etc/timezone has Asia/Shanghai

In Valetudo > Info it says Timezone: Asia/Singapore.

`date` command shows Singapore timezone (SGT).

Here it is described how you should configure timezone in Ubuntu (unattended): https://help.ubuntu.com/community/UbuntuTime#Using_the_Command_Line_.28unattended.29

After this, /etc/timezone and date are correct. But Valetudo > Info still shows Asia/Singapore. Where does this come from?

Hypfer, 2020-06-06:
This functionality actually just calls the `app_get_locale` and `set_timezone` miio command and returns its response.

Editing `/etc/timezone` doesn't do anything at all since the roborock software doesn't care about it because that's nothing that might happen without root.

What happens if you close and re-open the set timezone dialog in the timer settings? Apparently `get_timezone` and `app_get_locale` return different values.

I was able to set my timezone from `Europe/Berlin` to `Europe/Amsterdam` but the Info menu doesn't reflect that

schlamar, 2020-06-06:
Yes, the timezone info in timers is correct after an update.

The info menu reads data from /mnt/default/roborock.conf. However, this is mounted as read-only on my firmware.

As there is no way to check the current time in Valetudo, I would have expected that this is in sync with the linux time (result from `date`) and that is the way to make sure Valetudo has the correct time settings.

Hypfer, 2020-06-06:
The roborock blackbox which handles timers apparently doesn't care about the system timezone at all. Well.. apart from daily reboots I'd assume.

Honestly, 100% of the `app_get_locale` response is completely useless. It's probably time to remove it altogether

commit ffc06d65 (2020-06-06): fix: Removed useless app_get_locale functionality to fix #532
closed 2020-06-06 by Hypfer