Hypfer/Valetudo #536: embedded detection logic error?

rumpeltux, 2020-06-09:
## Describe the bug
embedded detection seems wrong for viomi and my local host

Logic in Model.js:

            try {
                fs.readFileSync("/etc/os-release");
                this.embedded = true;
            } catch (e) {
                this.embedded = false;
            }

So embedded is true if /etc/os-release is readable. This is the case on my local machine but not on the embedded robot. So the logic should be reversed? i.e.

    this.embedded = !fs.existsSync("/etc/os-release");

### Valetudo Version
HEAD

Hypfer, 2020-06-09:
You're completely right. I didn't think this through.

Any ideas how to properly solve this?

Simply inverting it isn't enough since there's no /etc/os-release on windows but there's one on roborock vacuums

rumpeltux, 2020-06-09:
For viomi `/mnt/UDISK` is a good marker.
I’m sure there’s one for roborock as well?

rumpeltux, 2020-06-09:
Or maybe something like `/etc/miio`?

Hypfer, 2020-06-09:
No `/etc/miio` or `/mnt/UDISK` on the roborock. I guess this might be more complicated than initially anticipated.

How does one autodetect whether or not it is running on a vacuum if the vacuum itself is basically a general purpose linux host.

I guess we could ditch embedded: auto and simply default to true since that will only be explicitly changed if it's not embedded.

rumpeltux, 2020-06-09:
We can have multiple checks. One for viomi, one for RR.

But defaulting to true also SGTM.

closed 2020-06-09 by Hypfer