Hypfer/Valetudo #539: RFE: add specific power mode per zone

sassmann, 2020-06-13:
First, thanks for this great piece of software!

I was wondering if it would be possible to specify a power mode per zone, so that for example a room with a carpet can run in turbo mode, while a room with a tiled floor may run in normal mode.
This option could just override the default setting in the "Home" menu during the zone cleaning.

Hypfer, 2020-06-13:
Newer roborock models have section-specific settings. I haven't looked into that yet.

We could probably simulate the same thing for other devices since we know where we are, but only if it's possible to do that properly and not just as a hack. 🤔

sassmann, 2020-06-13:
Hmm, don't think you need hardware support for that. If you save the default setting in Valetudo and apply the zone specific setting before starting the zone cleaning and revert it back to default (if necessary) after the zone is cleaned up.

Hypfer, 2020-06-13:
While it is surely nice to have features like this directly integrated into Valetudo so that you don't need to run something else, it's often a better idea to use a Smarthome system like openhab or home assistant or iobroker to do these advanced things.

That's both more flexible as well as maintainable.

That's also the reason why there will be no Telegram bot integration inside Valetudo. It's just better to do that in a separate component.

I'll have to think about what makes more sense here

sassmann, 2020-06-13:
Sure, you need to keep things easy and maintainable. From reading your comment you might have thought of it in a more sophisticated way than I had in mind.
If I read your first comment correctly you suggested to detect the location of the roborock and apply specific power modes based on that.
My idea is much more basic in the way that you would just apply the zone specific setting when manually selecting a zone for cleaning.

Either approach would be a cool feature, thanks for considering the idea!

Hypfer, 2020-06-13:
Ah! I see.

The issue with that approach would be to reliably detect when this zoned cleanup has ended.
I'm assuming that this could break when there's a new one started while the old one is active.

The main reason why I'm a bit sceptical with map-related stuff is that we're not the single source of truth. The robot might change the map without us noticing which would mean that the zones are completely off. There's no real way to do sanity-checking there I'm afraid.

Still, for the most part, we could get away with simply hoping that nothing will go wrong.

I'm hesitant to hardcode specific "If-this-than-that" conditions since people can and will have a lot of slightly different ideas.

Here are a few of those in RE:
https://github.com/rand256/valetudo/issues/93
https://github.com/rand256/valetudo/issues/85
https://github.com/rand256/valetudo/issues/16
https://github.com/rand256/valetudo/issues/50
https://github.com/rand256/valetudo/issues/44

While there all neat and all, I don't think that it is a good idea to reinvent the wheel inside Valetudo.

One Idea might be to add some of these commonly suggested Ideas including for example the necessary Home Assistant Automations to the documentation at [valetudo.cloud](https://valetudo.cloud)
That would be a lot easier to do than adding those to Valetudo itself while also being accessible to everyone.

If we're noticing that a feature is just so incredible and also extremely easy to do, we can then consider implementing it natively inside Valetudo. This would make this docs section a staging area for automation ideas.

Anyways..
So the easiest way to solve your use-case would be to implement #243 "Feature Request: Current Zone" which would mean that we'd simply check the current robot position against the stored zones and return all matches via MQTT.
This way, one could use any conditional action engine thing they like to change fan speeds etc.

sassmann, 2020-06-13:
Great explanation, thanks a lot.

closed 2021-01-09 by Hypfer