Hypfer/Valetudo #540: Roborock S6 - Can't enable persistent/lab mode

DomiStyle, 2020-06-14:
## Describe the bug
Trying to enable Persistent data on a Roborock S6 returns a 500 error from /api/set_lab_status

## To Reproduce
1. Go to Settings -> Persistent Data
2. Check "Enabled"
3. Click Save

### Vacuum Model
Roborock S6

### Valetudo Version
0.5.3

### Firmware Version
1708

### User-Agent
Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:78.0) Gecko/20100101 Firefox/78.0

## Expected behavior
Persistent data should be enabled but instead nothing happens.

Hypfer, 2020-06-15:
Could you check the response for its error code in the devtools network tab?

DomiStyle, 2020-06-15:
Unfortunately it only said ```[object Object]```.

I also tried enabling it manually with ```echo -n "1" > /mnt/data/rockrobo/lab.cfg``` but that completely kills mapping and only shows "No map data".
Since trying it manually I don't get a 500 anymore but just a timeout.

Hypfer, 2020-06-15:
Anything in `/var/log/upstart/valetudo.log`?

You can also increase log verbosity to `debug` or `trace` in ``/mnt/data/valetudo/config.json` and try again. That should log the raw messages exchanged with the robot

DomiStyle, 2020-06-15:
This is what comes up with debug logs enabled:

    [2020-06-15T08:27:07.971Z] [DEBUG] >>> local: {"method":"set_lab_status","params":[1],"id":17}
    [2020-06-15T08:27:07.977Z] [DEBUG] <<< local: {"error":{"code":-10005,"message":"First element in array is not an object"},"id":17}
    [2020-06-15T08:27:07.978Z] [INFO] error response {
    error: { code: -10005, message: 'First element in array is not an object' },
    id: 17
    }

Hypfer, 2020-06-15:
I see.

I'd assume that they've changed something about the command payload for the multi-floor support in `1708`

infinity312, 2020-07-16:
> I see.
> 
> I'd assume that they've changed something about the command payload for the multi-floor support in `1708`

Is there anything we can do to help with the fix/figure out what changed in 1708?

Hypfer, 2020-07-20:
Apparently, the lab status needs to be wrapped in an object like this: `{lab_status: 1}`

commit 19e141ef (2020-07-25): fix(vacuum.roborock): Changed set_lab_status payload for S6 to fix #540
closed 2020-07-25 by Hypfer