Hypfer/Valetudo #542: Rework MQTT Attributes

Hypfer, 2020-06-20:
The attributes mqtt topic has become crammed with things of very different nature.
It's also not very simple to work with attributes in home assistant in the first place.

Therefore, Valetudo should adopt the way it is done in Tasmota with entities for each attribute grouped by the device specification

Example:
[image]

Hypfer, 2020-07-27:
As mentioned in #568, this should also add the `sw_version` to the device entry

juantxorena, 2020-08-02:
I'm quite interested in controlling the robot via MQTT. I have no experience with MQTT (other than trying to use this), but I have done some REST APIs. Is there any way I can help with this task, maybe work on some schema for the MQTT API, or something?

duhow, 2020-10-19:
Adding a list of state attributes based in Roborock S6, added to Home Assistant via Xiaomi Miio:

```yaml
fan_speed_list:
  - Silent
  - Standard
  - Medium
  - Turbo
  - Gentle
battery_level: 100
battery_icon: 'mdi:battery-charging-100'
fan_speed: Turbo
do_not_disturb: 'on'
do_not_disturb_start: '22:00:00'
do_not_disturb_end: '08:00:00'
cleaning_time: 33
cleaned_area: 32
cleaning_count: 463
total_cleaned_area: 10580
total_cleaning_time: 10730
main_brush_left: 120
side_brush_left: 20
filter_left: 82
sensor_dirty_left: 13
status: Charging
clean_start: '2020-10-19T09:44:47'
clean_stop: '2020-10-19T10:18:32'
friendly_name: Roborock
supported_features: 14204
```

closed 2020-12-28 by Hypfer