Hypfer/Valetudo #545: Xiaomi Mijia 1S Support

sebiweise, 2020-06-22:
Has anyone got the Xiaomi Mijia 1S working yet? I tried my luck with the official firmware, but when building the Valetudo version I get the error that I don't have a gzip compatible archive.

Otherwise I would like to test it, if it works.

Hypfer, 2020-06-23:
What exactly are you doing?

sebiweise, 2020-06-23:
I followed the official instructions for the Roborocks and used the link in the description as image.
Unfortunately I can't get further than the error in the image build.

Hypfer, 2020-06-23:
I see. Those instructions only work for Gen1 and Gen2. We should clarify that.

While Valetudo technically supports the M1S, as of now, you will probably never get it running on it.

closed 2020-06-25 by sebiweise