Hypfer/Valetudo #548: Viomi STYTJ02YM viomi.vacuum.v8 support

Hypfer, 2020-06-25:
As reported(?) in #547, the current Viomi implementation may also work for the v8 model.

deku-m, 2020-06-26:
Is this the support also for  Xiaomi Mijia STYJ02YM V2 pro ?

rumpeltux, 2020-06-27:
> Is this support also for Xiaomi Mijia STYJ02YM V2 pro ?

The devices are probably similar enough, so likely: yes.

FTR: original report was https://github.com/rumpeltux/python-miio/issues/1#issuecomment-648451477

I’ll prepare a patch to add "official" support.

rumpeltux, 2020-06-27:
Also related support was added to python-miio two months ago: https://github.com/rytilahti/python-miio/issues/666

rumpeltux, 2020-06-27:
Added this to PR #543

closed 2020-06-28 by Hypfer

johnappletree, 2021-01-08:
First of all thanks for your work and contributions towards a cloud independent solution!

I tried installing version 2021.01.0b0 and 0.5.3 on Viomi STYTJ02YM viomi.vacuum.v8 my but both fail.

error message:
`<30>Jan  1 08:55:44 valetudo[713]: [1970-01-01T00:55:44.177Z] [INFO] Loading configuration file: /mnt/UDISK/config/valetudo.json
<30>Jan  1 08:55:44 valetudo[713]: [1970-01-01T00:55:44.216Z] [INFO] Starting Valetudo for Vacuum Model viomi.vacuum.v8
<30>Jan  1 08:55:44 valetudo[713]: [1970-01-01T00:55:44.217Z] [INFO] DeviceId 375361941
<30>Jan  1 08:55:44 valetudo[713]: [1970-01-01T00:55:44.218Z] [INFO] IP 127.0.0.1
<30>Jan  1 08:55:44 valetudo[713]: [1970-01-01T00:55:44.219Z] [INFO] CloudSecret xxxxxxxxxx
<30>Jan  1 08:55:44 valetudo[713]: [1970-01-01T00:55:44.222Z] [INFO] LocalSecret xxxxxxxxxx
<30>Jan  1 08:55:44 valetudo[713]: [1970-01-01T00:55:44.223Z] [INFO] JS Runtime Version v14.4.0
<30>Jan  1 08:55:44 valetudo[713]: [1970-01-01T00:55:44.224Z] [INFO] Max Heap Size: 33.5 MiB
<27>Jan  1 08:55:44 valetudo[713]: /snapshot/Valetudo/lib/Valetudo.js:59
<27>Jan  1 08:55:44 valetudo[713]:             throw new Error("No implementation found for " + this.model.getModelIdentifier());
<27>Jan  1 08:55:44 valetudo[713]:             ^
<27>Jan  1 08:55:44 valetudo[713]: 
<27>Jan  1 08:55:44 valetudo[713]: Error: No implementation found for viomi.vacuum.v8
<27>Jan  1 08:55:44 valetudo[713]:     at new Valetudo (/snapshot/Valetudo/lib/Valetudo.js:59:19)
<27>Jan  1 08:55:44 valetudo[713]:     at Object.<anonymous> (/snapshot/Valetudo/index.js:5:16)
<27>Jan  1 08:55:44 valetudo[713]:     at Module._compile (internal/modules/cjs/loader.js:1198:30)
<27>Jan  1 08:55:44 valetudo[713]:     at Module._compile (pkg/prelude/bootstrap.js:1327:32)
<27>Jan  1 08:55:44 valetudo[713]:     at Object.Module._extensions..js (internal/modules/cjs/loader.js:1218:10)
<27>Jan  1 08:55:44 valetudo[713]:     at Module.load (internal/modules/cjs/loader.js:1047:32)
<27>Jan  1 08:55:44 valetudo[713]:     at Function.Module._load (internal/modules/cjs/loader.js:935:14)
<27>Jan  1 08:55:44 valetudo[713]:     at Function.Module.runMain (pkg/prelude/bootstrap.js:1375:12)
<27>Jan  1 08:55:44 valetudo[713]:     at internal/main/run_main_module.js:17:47
`
Do you have any idea what causes this issue?
Best regards,
J.

rumpeltux, 2021-01-08:
@johnappletree You’ll need to install a newer version, e.g. >=0.6

johnappletree, 2021-01-08:
I tried the prerelease:  2021.01.0b0. But that did not work, version 0.6.1 seem to work. Thanks!

commit 17bb01ef (2020-06-28): feat(vacuum.viomi): add support for viomi.vacuum.v8 + mop enhancements (#543)

* Viomi: more mop-mode selection fixes

* Move 'New Status' diff log message from trace to debug verbosity

* Viomi: don't reset this.mode from vacuum status (which is unrelated). This requires to handle the case when initial status is default.

* Update Viomi installation instructions (fixes: #541)

* Viomi: Add a very basic knowledge base article (#538)

* Viomi: Only select mop-moves if mopping is enabled.

* Viomi: more debug logging for mopping state, fixes for startup mopping state detection.

* Add support fo [... cut]