Hypfer/Valetudo #550: No Maps for Viomi v8

jeste76, 2020-06-30:
## Describe the bug
On Viomi v8, the maps are not displayed.
## Description
I set up valetudo as described in the docs. In the webgui no maps are displayed. There is only a rectangle stating "No Map Data". 
I was not able to find any failed attemps to upload the mapdata in ``/tmp/``

## To Reproduce

Steps to reproduce the behavior:
1. Install valetudo as described in the docs
2. Open webgui
3. Navigate to Maps tab

### Screenshots

### Vacuum Model
viomi v8

### Valetudo Version
0.5.3

### Firmware Version
newest

### User-Agent
Firefox on linux / Firefox on iOS

## Expected behavior
Visible Maps

## Additional context
First I tried disguising the viomi v8 as  an viomi v8. After viomi v8 support was merged, I also tried that version.

rumpeltux, 2020-06-30:
To investigate this further we need debug logs. To do so enable logging per https://valetudo.cloud/pages/installation/viomi.html#enable-logging,
set the `"logLevel": "debug"` in `/mnt/data/valetudo/config.json` and reboot.

jeste76, 2020-07-01:
Thanks for the quick response. I will try to generate some logs as soon as I have some spare time.

jeste76, 2020-07-02:
I pasted some logs below. As far as I can see there is nothing about the maps. Can I somehow trigger a map upload attempt?

-----------
Mon Jan  5 12:43:18 1970 kern.err kernel: [    8.361969] android_usb: already disabled
Mon Jan  5 12:43:19 1970 kern.info kernel: [    8.466663] adb_open
Mon Jan  5 12:43:19 1970 kern.info kernel: [    8.466687] adb_bind_config
Mon Jan  5 12:43:19 1970 kern.warn kernel: [    8.466714] ep_matches, wrn: endpoint already claimed, ep(0xc0d5739c, 0xd79bf8c0, ep1in-bulk)
Mon Jan  5 12:43:19 1970 kern.warn kernel: [    8.907650] [ND]flush cache 1 0!
Mon Jan  5 12:43:21 1970 user.notice : device-mapper: remove ioctl on temporary-cryptsetup-546  failed: Resource busy
Mon Jan  5 12:43:21 1970 daemon.err block: Unkown action change
Mon Jan  5 12:43:21 1970 daemon.err block: Unkown action change
Mon Jan  5 12:43:22 1970 kern.info kernel: [   11.396486] EXT4-fs (dm-1): mounted filesystem with ordered data mode. Opts: (null)
Mon Jan  5 12:43:22 1970 user.notice : read_buf /dev/mapper/secret on /secret type ext4 (rw,relatime,data=ordered)
Mon Jan  5 12:43:22 1970 user.notice :  
Mon Jan  5 12:43:22 1970 user.notice : mount_secret_part succ  
Mon Jan  5 12:43:22 1970 daemon.err block: Unkown action change
Mon Jan  5 12:43:22 1970 kern.info kernel: [   12.233095] adb_release
Mon Jan  5 12:43:22 1970 kern.info kernel: [   12.233199] adb_open
Mon Jan  5 12:43:22 1970 kern.err kernel: [   12.259520] android_usb: already disabled
Mon Jan  5 12:43:23 1970 kern.warn kernel: [   12.468838] ****gareth drivers/rtc/rtc-sunxi.c:446 func sunxi_rtc_alarm_irq_enable() ***** 
Mon Jan  5 12:43:23 1970 kern.err kernel: [   13.051673] sndpcm_startup,l:1688,pa_vol:50
Mon Jan  5 12:43:23 1970 user.notice : 
Mon Jan  5 12:43:23 1970 user.notice : ----[CLogConfig::TConfigParams] ----
Mon Jan  5 12:43:23 1970 user.notice : 
Mon Jan  5 12:43:23 1970 user.notice : log_file_path                            = 
Mon Jan  5 12:43:23 1970 user.notice : log_directory_prefix                     = ./
Mon Jan  5 12:43:23 1970 user.notice : auto_build_directory                     = NO
Mon Jan  5 12:43:23 1970 user.notice : create_log_name_auto                     = NO
Mon Jan  5 12:43:23 1970 user.notice : log_time_string                          = YES
Mon Jan  5 12:43:23 1970 user.notice : log_module_string                        = YES
Mon Jan  5 12:43:23 1970 user.notice : log_username_string                      = YES
Mon Jan  5 12:43:23 1970 user.notice : log_level_string                         = YES
Mon Jan  5 12:43:23 1970 user.notice : also_printf                              = NO
Mon Jan  5 12:43:23 1970 user.notice : printf_this_call                         = YES
Mon Jan  5 12:43:23 1970 user.notice : add_to_logfile                           = YES
Mon Jan  5 12:43:23 1970 user.notice : different_log                            = YES
Mon Jan  5 12:43:23 1970 user.notice : out_log_level                            = 2
Mon Jan  5 12:43:23 1970 user.notice : out_log_type                             = 2
Mon Jan  5 12:43:23 1970 user.notice : log_user_list                            = 2047
Mon Jan  5 12:43:23 1970 user.notice : 
Mon Jan  5 12:43:23 1970 user.notice : CLog::init: file is /mnt/UDISK/log/miio_conf_check.temp for logging.
Mon Jan  5 12:43:23 1970 user.notice : CLog::init:  m_out_type:  None	 m_out_level:  LogNormal	 Logging User	 Logging Time	 Not also printing
Mon Jan  5 12:43:23 1970 user.notice : [CManagerInit] can not open log system perfix 
Mon Jan  5 12:43:24 1970 kern.warn kernel: [   13.540339] ==> rtl8188e_iol_efuse_patch
Mon Jan  5 12:43:24 1970 kern.warn kernel: [   13.775766] RTW: wlan0- hw port(0) mac_addr =5c:e5XXXXXXXXX
Mon Jan  5 12:43:24 1970 kern.warn kernel: [   13.775868] RTW: wlan1- hw port(1) mac_addr =5e:e5XXXXXXXXX
Mon Jan  5 12:43:24 1970 user.notice : ln: /lib/ld-musl-arm.so.1: File exists
Mon Jan  5 12:43:24 1970 daemon.info procd: - init complete -
Mon Jan  5 12:43:24 1970 kern.warn kernel: [   14.221986] [ND]flush cache 1 0!
Mon Jan  5 12:43:27 1970 kern.warn kernel: [   17.153802] RTW: rtw_set_802_11_connect(wlan0)  fw_state=0x00000000
Mon Jan  5 12:43:27 1970 kern.warn kernel: [   17.159212] RTW: start auth
Mon Jan  5 12:43:27 1970 kern.warn kernel: [   17.160839] RTW: auth success, start assoc
Mon Jan  5 12:43:27 1970 kern.warn kernel: [   17.174768] RTW: assoc success
Mon Jan  5 12:43:27 1970 kern.warn kernel: [   17.177602] RTW: ============ STA [34:2c:c4:0a:d1:ff]  ===================
Mon Jan  5 12:43:27 1970 kern.warn kernel: [   17.177618] RTW: mac_id : 0
Mon Jan  5 12:43:27 1970 kern.warn kernel: [   17.177625] RTW: wireless_mode : 0x0b
Mon Jan  5 12:43:27 1970 kern.warn kernel: [   17.177631] RTW: mimo_type : 0
Mon Jan  5 12:43:27 1970 kern.warn kernel: [   17.177638] RTW: bw_mode : 20MHz, ra_bw_mode : 20MHz
Mon Jan  5 12:43:27 1970 kern.warn kernel: [   17.177645] RTW: rate_id : 0
Mon Jan  5 12:43:27 1970 kern.warn kernel: [   17.177651] RTW: rssi : 52 (%), rssi_level : 0
Mon Jan  5 12:43:27 1970 kern.warn kernel: [   17.177659] RTW: is_support_sgi : Y, is_vht_enable : N
Mon Jan  5 12:43:27 1970 kern.warn kernel: [   17.177666] RTW: disable_ra : N, disable_pt : N
Mon Jan  5 12:43:27 1970 kern.warn kernel: [   17.177673] RTW: is_noisy : N
Mon Jan  5 12:43:27 1970 kern.warn kernel: [   17.177678] RTW: txrx_state : 0
Mon Jan  5 12:43:27 1970 kern.warn kernel: [   17.177685] RTW: curr_tx_rate : CCK_1M (L)
[... cut ...]
Mon Jan  5 13:08:30 1970 daemon.err valetudo[714]:     at EventEmitter.<anonymous> (/snapshot/Valetudo/lib/MqttClient.js:102:18)
Mon Jan  5 13:08:30 1970 daemon.err valetudo[714]:     at EventEmitter.emit (events.js:315:20)
Mon Jan  5 13:08:30 1970 daemon.err valetudo[714]:     at Events.emitStatusUpdated (/snapshot/Valetudo/lib/Events.js:24:27)
Mon Jan  5 13:08:30 1970 daemon.err valetudo[714]:     at Viomi.updateStatus (/snapshot/Valetudo/lib/devices/MiioVacuum.js:132:25)
Mon Jan  5 13:08:30 1970 daemon.err valetudo[714]:     at Viomi.pollStatus (/snapshot/Valetudo/lib/devices/Viomi.js:231:14)

rumpeltux, 2020-07-03:
Can you try to disable mqtt in `/mnt/data/valetudo/config.json`. It seems to be causing some errors. Not yet sure if related or not.
Also set `"logLevel": "trace"` so the log contains all the robot↔valetudo interactions.

jeste76, 2020-07-05:
The following logs were recorded with disabled mqtt and ``"loglevel": "trace"``. 
The Robo was charging at that time. I started the cleaning and aborted it after some seconds. 

<details><summary>Logs</summary>
<p>

```
Thu Jan  1 08:01:43 1970 user.notice : File descriptor 7 (pipe:[1278]) leaked on lvm invocation. Parent PID 424: /bin/sh
Thu Jan  1 08:01:43 1970 user.notice : File descriptor 9 (pipe:[2237]) leaked on lvm invocation. Parent PID 424: /bin/sh
Thu Jan  1 08:01:43 1970 user.notice :   Reading all physical volumes.  This may take a while...
Thu Jan  1 08:01:43 1970 kern.warn kernel: [    6.554700] [NAND]BLKFLSBUF called!
Thu Jan  1 08:01:43 1970 kern.warn kernel: [    6.555250] [ND]flush cache 1 0!
Thu Jan  1 08:01:43 1970 kern.warn kernel: [    6.555322] [NAND]BLKFLSBUF called!
Thu Jan  1 08:01:43 1970 kern.warn kernel: [    6.555408] [NAND]BLKFLSBUF called!
Thu Jan  1 08:01:43 1970 kern.warn kernel: [    6.556694] [NAND]BLKFLSBUF called!
Thu Jan  1 08:01:43 1970 user.notice : File descriptor 7 (pipe:[1278]) leaked on lvm invocation. Parent PID 424: /bin/sh
Thu Jan  1 08:01:43 1970 user.notice : File descriptor 9 (pipe:[2237]) leaked on lvm invocation. Parent PID 424: /bin/sh
Thu Jan  1 08:01:43 1970 kern.warn kernel: [    6.594520] [NAND]BLKFLSBUF called!
Thu Jan  1 08:01:43 1970 kern.warn kernel: [    6.595724] [ND]flush cache 1 0!
Thu Jan  1 08:01:43 1970 kern.warn kernel: [    6.595803] [NAND]BLKFLSBUF called!
Thu Jan  1 08:01:43 1970 kern.warn kernel: [    6.595898] [NAND]BLKFLSBUF called!
Thu Jan  1 08:01:43 1970 kern.warn kernel: [    6.595993] [NAND]BLKFLSBUF called!
Thu Jan  1 08:01:43 1970 user.notice : this file has been obsoleted. please call "/sbin/block mount" directly
Thu Jan  1 08:01:43 1970 daemon.err block: /dev/mapper/rootfs is already mounted
Thu Jan  1 08:01:43 1970 daemon.notice netifd: Interface 'loopback' is enabled
Thu Jan  1 08:01:43 1970 daemon.notice netifd: Interface 'loopback' is setting up now
Thu Jan  1 08:01:43 1970 daemon.notice netifd: Interface 'loopback' is now up
Thu Jan  1 08:01:43 1970 daemon.notice netifd: Network device 'lo' link is up
Thu Jan  1 08:01:43 1970 daemon.notice netifd: Interface 'loopback' has link connectivity 
Thu Jan  1 08:01:43 1970 kern.info kernel: [    6.905860] EXT4-fs (nandk): recovery complete
Thu Jan  1 08:01:43 1970 kern.info kernel: [    6.906565] EXT4-fs (nandk): mounted filesystem with ordered data mode. Opts: 
Thu Jan  1 08:01:43 1970 daemon.err block: /dev/by-name/rootfs is already mounted
Thu Jan  1 08:01:43 1970 daemon.err block: /dev/by-name/rootfs_data is already mounted
Thu Jan  1 08:01:43 1970 authpriv.info dropbear[498]: Not backgrounding
Thu Jan  1 08:01:43 1970 kern.err kernel: [    7.249915] android_usb: already disabled
Thu Jan  1 08:01:44 1970 kern.info kernel: [    7.354014] adb_open
Thu Jan  1 08:01:44 1970 kern.info kernel: [    7.354037] adb_bind_config
Thu Jan  1 08:01:44 1970 kern.warn kernel: [    7.354062] ep_matches, wrn: endpoint already claimed, ep(0xc0d5739c, 0xd79a7180, ep1in-bulk)
Thu Jan  1 08:01:44 1970 kern.warn kernel: [    7.792578] [ND]flush cache 1 0!
Thu Jan  1 08:01:46 1970 user.notice : device-mapper: remove ioctl on temporary-cryptsetup-545  failed: Resource busy
Thu Jan  1 08:01:46 1970 daemon.err block: Unkown action change
Thu Jan  1 08:01:46 1970 daemon.err block: Unkown action change
Thu Jan  1 08:01:47 1970 user.notice : read_buf /dev/mapper/secret on /secret type ext4 (rw,relatime,data=ordered)
Thu Jan  1 08:01:47 1970 user.notice :  
Thu Jan  1 08:01:47 1970 user.notice : mount_secret_part succ  
Thu Jan  1 08:01:47 1970 kern.info kernel: [   10.300606] EXT4-fs (dm-1): mounted filesystem with ordered data mode. Opts: (null)
Thu Jan  1 08:01:47 1970 daemon.err block: Unkown action change
Thu Jan  1 08:01:47 1970 user.notice : Changing password for root
Thu Jan  1 08:01:47 1970 user.notice : New password: 
Thu Jan  1 08:01:47 1970 kern.info kernel: [   11.093180] adb_release
Thu Jan  1 08:01:47 1970 kern.info kernel: [   11.093275] adb_open
Thu Jan  1 08:01:47 1970 kern.err kernel: [   11.123648] android_usb: already disabled
Thu Jan  1 08:01:48 1970 kern.warn kernel: [   11.353078] ****gareth drivers/rtc/rtc-sunxi.c:446 func sunxi_rtc_alarm_irq_enable() ***** 
Thu Jan  1 08:01:48 1970 user.notice : Retype password: 
Thu Jan  1 08:01:48 1970 auth.info passwd: Password for root changed by root
Thu Jan  1 08:01:48 1970 user.notice : Password for root changed by root
Thu Jan  1 08:01:48 1970 kern.err kernel: [   11.958864] sndpcm_startup,l:1688,pa_vol:50
Thu Jan  1 08:01:48 1970 user.notice : 
Thu Jan  1 08:01:48 1970 user.notice : ----[CLogConfig::TConfigParams] ----
Thu Jan  1 08:01:48 1970 user.notice : 
Thu Jan  1 08:01:48 1970 user.notice : log_file_path                            = 
Thu Jan  1 08:01:48 1970 user.notice : log_directory_prefix                     = ./
Thu Jan  1 08:01:48 1970 user.notice : auto_build_directory                     = NO
Thu Jan  1 08:01:48 1970 user.notice : create_log_name_auto                     = NO
Thu Jan  1 08:01:48 1970 user.notice : log_time_string                          = YES
Thu Jan  1 08:01:48 1970 user.notice : log_module_string                        = YES
Thu Jan  1 08:01:48 1970 user.notice : log_username_string                      = YES
Thu Jan  1 08:01:48 1970 user.notice : log_level_string                         = YES
Thu Jan  1 08:01:48 1970 user.notice : also_printf                              = NO
[... cut ...]
Thu Jan  1 08:36:53 1970 daemon.info valetudo[710]: [1970-01-01T00:36:53.149Z] [TRACE] New Status: {
Thu Jan  1 08:36:53 1970 daemon.info valetudo[710]:   state: 'CHARGING',
Thu Jan  1 08:36:53 1970 daemon.info valetudo[710]:   error_code: 'CHARGING',
Thu Jan  1 08:36:53 1970 daemon.info valetudo[710]:   human_state: 'CHARGING',
Thu Jan  1 08:36:53 1970 daemon.info valetudo[710]:   'viomi.vacuum.v8_run_state': 5,
Thu Jan  1 08:36:53 1970 daemon.info valetudo[710]:   'viomi.vacuum.v8_err_state': 2103
Thu Jan  1 08:36:53 1970 daemon.info valetudo[710]: }

```

</p>
</details>

rumpeltux, 2020-07-07:
From your logs I can see that Valetudo is only using the local-lan interface to talk to the robot service.
To support maps though it needs to have connectivity via the cloud interface. (Logs should show `[INFO] Cloud connected` in that case soon after Valetudo started).

Can you share your robots `/etc/hosts` and the the output of `iptables -L -n -t nat`?

jeste76, 2020-07-07:
Sure, here you go:

```
root@TinaLinux:~# cat /etc/hosts
127.0.0.1 localhost

::1     localhost ip6-localhost ip6-loopback
ff02::1 ip6-allnodes
ff02::2 ip6-allrouters
110.43.0.83 ot.io.mi.com ott.io.mi.com
110.43.0.83 ot.io.mi.com ott.io.mi.com
110.43.0.83 de.ot.io.mi.com de.ott.io.mi.com
110.43.0.83 ea.ot.io.mi.com ea.ott.io.mi.com
110.43.0.83 in.ot.io.mi.com in.ott.io.mi.com
110.43.0.83 pv.ot.io.mi.com pv.ott.io.mi.com
110.43.0.83 ru.ot.io.mi.com ru.ott.io.mi.com
110.43.0.83 sg.ot.io.mi.com sg.ott.io.mi.com
110.43.0.83 st.ot.io.mi.com st.ott.io.mi.com
110.43.0.83 tw.ot.io.mi.com tw.ott.io.mi.com
110.43.0.83 us.ot.io.mi.com us.ott.io.mi.com
```
```
root@TinaLinux:~# iptables -L -n -t nat
Chain PREROUTING (policy ACCEPT)
target     prot opt source               destination         

Chain INPUT (policy ACCEPT)
target     prot opt source               destination         

Chain OUTPUT (policy ACCEPT)
target     prot opt source               destination         
DNAT       tcp  --  0.0.0.0/0            110.43.0.83          tcp dpt:80 to:192.168.0.150:8080
DNAT       udp  --  0.0.0.0/0            110.43.0.83          udp dpt:8053 to:192.168.0.150:8053
DNAT       tcp  --  0.0.0.0/0            110.43.0.85          tcp dpt:80 to:192.168.0.150:8080
DNAT       udp  --  0.0.0.0/0            110.43.0.85          udp dpt:8053 to:192.168.0.150:8053

Chain POSTROUTING (policy ACCEPT)
target     prot opt source               destination  
```
192.168.0.150 is the ip of the vacuum.

rumpeltux, 2020-07-11:
Can you first try to change `192.168.0.150` to `127.0.0.1` in `/etc/rc.d/S51valetudo`? (that’s what mine is using)
Then check if you see the "Cloud connected" in the log after a reboot.

It could also be that robot is connecting to a hostname not covered in the list. You can try to log the domain names that the robot tries to resolve.

    tcpdump -nnvvXSs 1500 -i wlan0 port 53

Probably need to do this before the robot app starts, so do

    rm /etc/rc.d/S90robotManager

Then reboot. Then run the tcpdump command in one terminal, and then run the following in another:

    /etc/init.d/robotManager start
    cd /etc/rc.d
    ln -s ../init.d/robotManager S90robotManager

jeste76, 2020-07-13:
After changing the IP to ``127.0.0.1`` in ``/etc/rc.d/S51valetudo`` the "Cloud connected" finally shows up in the logs. There are  some logs regarding map upload now. But sadly the are not displayed in the web-ui. 

In the logs, these blocks show up:
```
Wed Jan  7 06:19:38 1970 daemon.info valetudo[715]: [1970-01-06T22:19:38.181Z] [DEBUG] <<< cloud: {"id":134,"method":"_sync.gen_tmp_presigned_url","params":{"index":[0]}}
Wed Jan  7 06:19:38 1970 daemon.info valetudo[715]: [1970-01-06T22:19:38.183Z] [DEBUG] >>> cloud: {"id":134,"result":{"urls":[{"url":"http://127.0.0.1/api/miio/map_upload_handler?ts=173,740404544&suffix=urls","method":"PUT"},{"url":"http://127.0.0.1/api/miio/map_upload_handler?ts=173,740520794&suffix=urls","method":"PUT"},{"url":"http://127.0.0.1/api/miio/map_upload_handler?ts=173,740555294&suffix=urls","method":"PUT"},{"url":"http://127.0.0.1/api/miio/map_upload_handler?ts=173,740577252&suffix=urls","methWed Jan  7 06:19:43 1970 daemon.info valetudo[715]: [1970-01-06T22:19:43.385Z] [INFO] Get FanSpeeds
Wed Jan  7 06:19:48 1970 daemon.info valetudo[715]: [1970-01-06T22:19:48.201Z] [DEBUG] <<< cloud: {"id":137,"method":"_sync.gen_tmp_presigned_url","params":{"index":[0]}}
Wed Jan  7 06:19:48 1970 daemon.info valetudo[715]: [1970-01-06T22:19:48.203Z] [DEBUG] >>> cloud: {"id":137,"result":{"urls":[{"url":"http://127.0.0.1/api/miio/map_upload_handler?ts=183,760188381&suffix=urls","method":"PUT"},{"url":"http://127.0.0.1/api/miio/map_upload_handler?ts=183,760294089&suffix=urls","method":"PUT"},{"url":"http://127.0.0.1/api/miio/map_upload_handler?ts=183,760323381&suffix=urls","method":"PUT"},{"url":"http://127.0.0.1/api/miio/map_upload_handler?ts=183,760349839&suffix=urls","methWed Jan  7 06:19:49 1970 daemon.info valetudo[715]: [1970-01-06T22:19:49.862Z] [INFO] Get FanSpeeds
Wed Jan  7 06:19:54 1970 daemon.info valetudo[715]: [1970-01-06T22:19:54.075Z] [INFO] Get FanSpeeds
Wed Jan  7 06:19:56 1970 daemon.info valetudo[715]: [1970-01-06T22:19:56.964Z] [DEBUG] <<< cloud* {"stamp":512396}
```

I guess, singe the "cloud connected" is in the logs, there is no need to do the tcpdump anymore? Or are there additional server, regarding the maps?

rumpeltux, 2020-07-13:
No need for tcpdump then.

If your loglevel is still `trace` you should also see some attempts of map-parsing in the logs.
Are there any mapdata files in /tmp on the robot? `ls /tmp/mapdata*`

If yes, then map-parsing failed. In that case I need a sample to look at. These maps contain room names & layout as well as robot path, so you may want to share this privately. You can find my email in the commit logs.

jeste76, 2020-07-13:
Sadly, there is no occurrence of the phrase "pars" in the logs (three days now). The loglevel is still set to ``trace`` The folder ``/tmp/mapdata`` has not been created yet.

rumpeltux, 2020-07-15:
The messages should look like this:

```<30>Jun 22 01:12:56 valetudo[2199]: [2020-06-21T17:12:56.547Z] [DEBUG] map_upload_handler { ts: '31,547684058', suffix: 'urls' }
<30>Jun 22 01:12:56 valetudo[2199]: [2020-06-21T17:12:56.565Z] [DEBUG] <<< cloud: {"result":["ok"],"id":154}
<30>Jun 22 01:12:56 valetudo[2199]: [2020-06-21T17:12:56.566Z] [INFO] << cloud: ignoring response for non-pending request {"result":["ok"],"id":154}
<30>Jun 22 01:12:56 valetudo[2199]: [2020-06-21T17:12:56.587Z] [TRACE] take 4@mask 0 <Buffer ff 78 00 00>
<30>Jun 22 01:12:56 valetudo[2199]: [2020-06-21T17:12:56.587Z] [TRACE] take 44@robot status 4 <Buffer b4 d2 63 5e 01 00 00 00 0a 00 00 00 c3 00 00 00 00 00 00 00 03 00 00 00 36 08 00 00 00 00 00 00 00 00 00 00 03 00 00 00 60 01 00 00>
``` 

/tmp/mapdata isn’t a folder but there should be files like /tmp/mapdata0. If those don’t exist and you don’t see respective lines in the log then the robot isn’t sending maps (or fails to connect or so). Not sure how to debug further then.

jeste76, 2020-07-29:
I just wanted to give a quick update on this:

Sadly I did not have much time to play around with the robot. I updated to the newest valetudo version. However, there is still no sign of map parsing in the logs or any mapdata in the /tmp folder. 

Do I have to make some settings in the original Xiaomi app beforehand? Valetudo was installed pretty much on an untouched robot, in my case. Is there a way to poll maps from the robot without fully running valetudo/ via shell? 

Do you know whether other people have the viomi v8 running with maps support?

waaaghs, 2020-08-05:
Same issue for me
New robot (v8) everything but maps working so far

I tried everthing from above and have a quite similar behaviour: I also get a "Cloud connected" after switching to 127.0.0.1 but no files in /tmp/mapdata*

PaskoMoto, 2020-08-10:
Hi.

Just in case this helps:

I had the same problem because I configured webserver port with 80.... changed it to 8080 in /mnt/UDISK/config/valetudo.json. Later I left the vacuum clean the hole apartment and when it finished a complete map was available for the first time. 

What I have noticed however, is that the map does not last after reboot, althoug I'm pretty sure it stands in /mn/UDISK/log/0_png.png 

I would say that even without traces in /tmp the maps is being uploaded properly..... just my 2 cents.

Best Regards.
PaskoMoto.

jwiesel, 2020-08-12:
Hi there,
unfortunately I have a similar issue with my Roborock S50. The map always says "no map data".
Workaround approach:
Disabling MQTT, activating persisted map and rebooting the device via the hardware button (turn off and turn on) made the live-map at least appear for me while cleaning.

On the other hand there is still no folder /tmp/mapdata, but just various entries with four letters like "ApLg", "McCa"
The logs still say things like:
```[2020-08-12T11:00:01.330Z] [INFO] Webserver running on port 80
[2020-08-12T11:00:45.246Z] [INFO] Cloud connected
[2020-08-12T11:01:26.108Z] [WARN] Failed to parse uploaded map
[2020-08-12T11:01:28.119Z] [WARN] Failed to parse uploaded map
[2020-08-12T11:01:32.167Z] [WARN] Failed to parse uploaded map
[2020-08-12T11:06:08.882Z] [ERROR] unhandledRejection MiioTimeoutError: request timed out:{"method":"get_status","params":{},"id":280}
    at Timeout.timeout [as _onTimeout] (/snapshot/Valetudo/lib/miio/MiioSocket.js:168:32)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7)
[2020-08-12T11:06:12.885Z] [ERROR] unhandledRejection MiioTimeoutError: request timed out:{"method":"get_status","params":{},"id":281}
    at Timeout.timeout [as _onTimeout] (/snapshot/Valetudo/lib/miio/MiioSocket.js:168:32)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7)
[2020-08-12T11:06:16.897Z] [ERROR] unhandledRejection MiioTimeoutError: request timed out:{"method":"get_status","params":{},"id":282}
    at Timeout.timeout [as _onTimeout] (/snapshot/Valetudo/lib/miio/MiioSocket.js:168:32)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7)
[2020-08-12T11:06:20.905Z] [ERROR] unhandledRejection MiioTimeoutError: request timed out:{"method":"get_status","params":{},"id":283}
    at Timeout.timeout [as _onTimeout] (/snapshot/Valetudo/lib/miio/MiioSocket.js:168:32)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7)
[2020-08-12T11:06:24.914Z] [ERROR] unhandledRejection MiioTimeoutError: request timed out:{"method":"get_status","params":{},"id":284}
    at Timeout.timeout [as _onTimeout] (/snapshot/Valetudo/lib/miio/MiioSocket.js:168:32)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7)
```

My system:
Pre-built firmware image taken from: https://github.com/zvldz/vacuum --> https://vacuumz.info/download/gen2/vacuum_2020_valetudo_0_6_1.pkg

Hypfer, 2020-08-12:
@jwiesel 

>Pre-built firmware image
Please don't do that

>On the other hand there is still no folder /tmp/mapdata
That's something specific to the Viomi implementation

>Failed to parse uploaded map
Interesting. I've never seen that before 🤔

Still, this issue is about Viomi and not Roborock so this conversation should be moved elsewhere

rumpeltux, 2020-08-17:
> I had the same problem because I configured webserver port with 80.... changed it to 8080 in /mnt/UDISK/config/valetudo.json. Later I left the vacuum clean the hole apartment and when it finished a complete map was available for the first time.

Hmm, this does make sense. Viomi is missing proper installation instructions. https://valetudo.cloud/pages/installation/viomi.html#robot-setup assumes that valetudo is running on a different host and thus the iptables redirect to port 8080.

Just to confirm: you were able to see the map in the valetudo UI? That would mean in your scenario map upload and parsing was working correctly.

The robot should however persist maps across reboots which is independent of valetudo. Maybe there is another command required to persist a new temporary map (which gets created as result of cleaning). The official app has a dialog asking whether to keep a new map.

jeste76, 2020-08-18:
Yesterday I found some time and reinstalled Valetudo on my Viomi. While doign so, I also changed my port to 8080.
The maps are now visible in the Web UI and they even seem to persist after a reboot.

PaskoMoto, 2020-08-19:
> Hmm, this does make sense. Viomi is missing proper installation instructions. https://valetudo.cloud/pages/installation/viomi.html#robot-setup assumes that valetudo is running on a different host [...]

That's it. Problem now is that I can't seem to make it persist after reboot... or even between two consecutive cleaning processes.

> The robot should however persist maps across reboots which is independent of valetudo. Maybe there is another command required to persist a new temporary map (which gets created as result of cleaning). The official app has a dialog asking whether to keep a new map.

That's what I've been thinking. For some days a map of a single room persisted after reboots, even after new maps were generated while cleaning the while apartment later. 
Now that you say that maps are independent from valetudo I think I'll try a new clean of the whole apartment without valetudo running, and see if I can definitely get a persistent map.

Best Regards.
PaskoMoto.

PaskoMoto, 2020-08-24:
Hi.
I'm still struggling with this issue. So I decided to activate the low level traces and paste them here. I get an OutOfBounds error:

`[2020-08-24T19:41:12.050Z] [DEBUG] <<< cloud: {"result":["ok"],"id":1}
[2020-08-24T19:41:12.051Z] [INFO] << cloud: ignoring response for non-pending request {"result":["ok"],"id":1}
[2020-08-24T19:41:12.081Z] [TRACE] take 2@featureFlags 0 <Buffer ff 78>
[2020-08-24T19:41:12.085Z] [TRACE] take 2@unknown01 2 <Buffer 00 00>
[2020-08-24T19:41:12.090Z] [TRACE] take 44@robot status 4 <Buffer 00 00 00 00 00 00 00 00 00 00 00 00 c5 00 00 00 01 00 00 00 03 00 00 00 37 08 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00>
[2020-08-24T19:41:12.092Z] [TRACE] take 40@map head 30 <Buffer 00 00 00 00 01 00 00 00 01 00 00 00 20 03 00 00 20 03 00 00 00 00 20 c1 00 00 20 c1 00 00 20 41 00 00 20 41 cd cc cc 3c>
[2020-08-24T19:41:12.164Z] [TRACE] take 12@history 9c458 <Buffer 00 00 00 00 00 00 00 00 00 00 00 00>
[2020-08-24T19:41:12.166Z] [TRACE] take 16@charge station 9c464 <Buffer 00 00 00 00 00 80 89 44 00 80 89 44 00 00 00 00>
[2020-08-24T19:41:12.169Z] [TRACE] take 12@virtual wall 9c474 <Buffer 00 00 00 00 00 00 00 00 00 00 00 00>
[2020-08-24T19:41:12.170Z] [TRACE] VirtualWall [ 0, 0, 0 ]
[2020-08-24T19:41:12.172Z] [TRACE] take 0@virtual wall content 9c480 <Buffer >
[2020-08-24T19:41:12.173Z] [TRACE] take 12@area head 9c480 <Buffer 00 00 00 00 01 00 00 00 00 00 00 00>
[2020-08-24T19:41:12.174Z] [TRACE] AreaClean [ 0, 1, 0 ]
[2020-08-24T19:41:12.175Z] [TRACE] take 0@area content 9c48c <Buffer >
[2020-08-24T19:41:12.176Z] [TRACE] take 20@navigate 9c48c <Buffer 00 00 00 00 00 00 00 00 00 80 89 44 00 80 89 44 00 00 00 00>
[2020-08-24T19:41:12.177Z] [TRACE] take 21@realtime 9c4a0 <Buffer 00 00 00 00 00 00 00 00 00 33 7b 89 44 00 80 89 44 00 00 00 00>
[2020-08-24T19:41:12.178Z] [TRACE] take 8@unknown8 9c4b5 <Buffer 00 00 00 00 00 00 00 00>
[2020-08-24T19:41:12.180Z] [TRACE] take 4@room header 9c4bd <Buffer 00 00 00 00>
[2020-08-24T19:41:12.181Z] [TRACE] take 4@mapArg: 9c4c2 <Buffer 00 00 00 00>
[2020-08-24T19:41:12.182Z] [TRACE] take 4@room length 9c4c6 <Buffer 00 00 00 00>
[2020-08-24T19:41:12.184Z] [TRACE] take 6@unknown 9c4ca <Buffer 00 72 fe 00 0c 70>
[2020-08-24T19:41:12.185Z] [TRACE] take 50@unknown50 9c4d0 <Buffer fe 00 60 12 88 b5 b9 4c ef b6 02 00 00 00 c8 72 f0 b6 28 00 00 00 d4 76 f0 b6 02 00 00 00 c8 72 f0 b6 28 00 00 00 d4 76 f0 b6 c8 72 f0 00 00 00 00>
[2020-08-24T19:41:12.185Z] [TRACE] take 5@unknown5 9c502 <Buffer >
[2020-08-24T19:41:12.187Z] [TRACE] take 4@pose length 9c507 <Buffer >
[2020-08-24T19:41:12.194Z] [WARN] Unable to parse Pose RangeError [ERR_BUFFER_OUT_OF_BOUNDS]: Attempt to access memory outside buffer bounds
    at boundsError (internal/buffer.js:75:11)
    at Buffer.readUInt32LE (internal/buffer.js:211:5)
    at ViomiMapParser.parsePose (/snapshot/Valetudo/lib/ViomiMapParser.js:270:50)
    at ViomiMapParser.parse (/snapshot/Valetudo/lib/ViomiMapParser.js:137:18)
    at Viomi.parseMap (/snapshot/Valetudo/lib/devices/Viomi.js:451:50)
    at /snapshot/Valetudo/lib/webserver/DeviceRouter.js:31:50
    at processTicksAndRejections (internal/process/task_queues.js:97:5) {
  code: 'ERR_BUFFER_OUT_OF_BOUNDS'
}
`
Best Regards.

Hypfer, 2020-08-24:
What version is this @PaskoMoto ?

PaskoMoto, 2020-08-24:
Hi.

This is 0.6.1.

Best Regards.

rumpeltux, 2020-08-26:
Looks like there’s some parsing error. Could you share a map dump from the robots `/tmp` directory? E.g. `/tmp/mapdata0`?

PaskoMoto, 2020-08-31:
> Looks like there’s some parsing error. Could you share a map dump from the robots `/tmp` directory? E.g. `/tmp/mapdata0`?

Hi.
There's no directory called `mapdata0` under `/tmp`
However, there's always an `0_png.png` file in `/mnt/UDISK/logs `with a map of the last cleaning process but I'm not sure whether it's generated by valetudo or the original vacuum software.
Sorry for late answer.
Best Regards.
PaskoMoto.

PaskoMoto, 2020-09-02:
Hi.

Here's a capture of a complete map. No dumps that I am aware of. 

http://imgfz.com/i/LmViXjq.png

By the way: What is the meaning of all those grey triangles around the map?

Best Regards.
PaskoMoto.

rumpeltux, 2020-09-04:
If you see a map in the valetudo UI it has been correctly parsed. Only in the event of parse failures should dumps be created in `/tmp/`, but these do get remove on reboots.

PaskoMoto, 2020-09-04:
> If you see a map in the valetudo UI it has been correctly parsed. Only in the event of parse failures should dumps be created in `/tmp/`, but these do get remove on reboots.

Hi.

Ah, ok. Now I understand.
I hadn't rebooted the device yet because I fear losing the maps again.
Should the map persist after a reboot?
And the last question: What do the grey triangles mean?

Best regards.

codingstube, 2020-09-18:
Hi everyone,

does valetudo parse the map from the xiaomi app? Or do I have to clean with valetudo to get the first map?

rumpeltux, 2020-09-18:
> does valetudo parse the map from the xiaomi app? Or do I have to clean with valetudo to get the first map?

The map is stored on the robot. Valetudo reads it and parses it.
Maybe xiaomi cloud stores a copy, but Valetudo does not access it.

You should see a map immediately after starting cleaning, if you don’t something is broken.

codingstube, 2020-09-18:
Ok I'll give it a try. Do I have to change the webserver port to 8080 as mentioned above?

rumpeltux, 2020-09-18:
> Ok I'll give it a try. Do I have to change the webserver port to 8080 as mentioned above?

Here are some updated docs: https://github.com/rumpeltux/Valetudo/blob/master/docs/_pages/installation/viomi.md#robot-setup
Make sure your config is matching. So if your S51valetudo points to 8080, your config should too.

codingstube, 2020-09-19:
Thanks.
I got it working. The map and status is showing up correctly.

But if I select a room and click the Start Button. A message appears at the bottom "unidentified".
If I start the viaomi via home page it's start cleaning.

I do not really see anything wrong in the log.
`1970-01-05T04:46:16.517Z] [INFO] << cloud: ignoring response for non-pending request {"result":["ok"],"id":8}
[1970-01-05T04:46:30.764Z] [DEBUG] <<< cloud* {"stamp":362790}
[1970-01-05T04:46:30.767Z] [DEBUG] >>> cloud* {"stamp":362790}
[1970-01-05T04:46:30.812Z] [DEBUG] <<< cloud: {"partner_id":"","params":{"hw_ver":"Linux","fw_ver":"3.5.3_0017","ap":{"ssid":"xxxxxxx","bssid":"82:2a:a8:eb:cf:3a"},"netif":{"localIp":"192.168.0.123","mask":"255.255.255.0","gw":"192.168.0.1"},"model":"viomi.vacuum.v8","mac":"xxxxxxxxxxxxx","token":"xxxxxxxxx","life":1134},"id":1292018947,"method":"_otc.info"}
[1970-01-05T04:46:30.814Z] [DEBUG] >>> cloud: {"id":1292018947,"result":{"otc_list":[{"ip":"203.0.113.1","port":8053}],"otc_test":{"list":[{"ip":"203.0.113.1","port":8053}],"interval":1800,"firsttest":1193}}}
[1970-01-05T04:46:39.909Z] [TRACE] >>> cloud: {"method":"get_prop","params":["run_state","mode","err_state","battary_life","box_type","mop_type","s_time","s_area","suction_grade","water_grade","remember_map","has_map","is_mop","has_newmap"],"id":9}
[1970-01-05T04:46:39.948Z] [TRACE] <<< cloud: {"result":[1,0,0,98,1,0,0,0,0,12,1,1,0,0],"id":9}
[1970-01-05T04:46:50.764Z] [DEBUG] <<< cloud* {"stamp":362810}
[1970-01-05T04:46:50.767Z] [DEBUG] >>> cloud* {"stamp":362810}
`

I get warnings of parsing the maps

schannall, 2020-10-05:
Does someone else still have the issue of the ticket ("No Maps vor Viomi V8", _not_ persistent maps after reboot)? This seems to be resolved for the most people? As such this Issue could be marked "solved".

That being said: I don't have a problem with Maps after a reboot.

Things to look out for:

1.   in `/mnt/data/valetudo/config.json` look out for

``` "map_upload_host": "http://127.0.0.1",```

2. in `/mnt/data/valetudo/config.json` look out for  
```
"dummycloud": {
    "spoofedIP": "110.43.0.83",
    "bindIP": "127.0.0.1"
```
3. in `/etc/rc.d/S51valetudo`:
```
dest=127.0.0.1
port=80
```
(Some People had success with changing the Port to 8080)

PaskoMoto, 2020-10-05:
Hi.

As far as I'm concerned, this issue can be marked as solved. 

Best Regards.
PaskoMoto.

petec999, 2020-10-06:
I've got some concern:
you are writing
Check your config under: /mnt/data/valetudo/config.json

I have that part in:
/mnt/UDISK/config/valetudo.json

is it ok?

schannall, 2020-10-06:
@petec999 :
do you use a v7 or an v8? AFAIK the v7 has the config under /mnt/UDISK
Also: are you on 0.6.1?

In the end: When starting Valetudo does it load the config under /mnt/UDISK/ or under /mnt/data/...?

petec999, 2020-10-06:
V8, no /mnt/data, 0.6.1

schannall, 2020-10-06:
Seems to be alright.
What do you have set your "map_upload_host" to?

PaskoMoto, 2020-10-06:
Hi.

I have an v8 devide.
Config file is located under:

`/mnt/UDISK/config/`

Best Regards.
PaskoMoto.

PaskoMoto, 2020-10-06:
Hi.

I own an v8 device.

Config file is located under:

`/mnt/UDISK/config/`

and works fine.

Best Regards.
PaskoMoto.

petec999, 2020-10-06:
`  "map_upload_host": "http://127.0.0.1",`

oot: how to change lang to polish, I have it in /mnt/UDISK/Audio/pl, 
mandarin and English are in /usr/share/audio (omitting replacing english which I have now by polish)...

schannall, 2020-10-06:
@petec999 :
what do you have you bindIP set to?
    "bindIP": "127.0.0.1"

petec999, 2020-10-06:
"dummycloud": {
    "spoofedIP": "203.0.113.1",
    "bindIP": "127.0.0.1"

schannall, 2020-10-06:
Why do you have that Spoofed IP?!
You natted 110.43.0.83...
Also - what does your /etc/hosts say?

petec999, 2020-10-06:
I don't know where from I got this spoofed IP :-)
I have just repeated procedure, step by step, and now I have map, on port 80.
My question is: I have 2 floors, and I want to upload appropriate map before cleaning.
Which file I should preserve for replacing?
Maybe I will play with samba for replacing appropriate file....

petec999, 2020-10-07:
does map stand in any particular file? If it is 'uploaded' I think yes ;)

schannall, 2020-10-07:
Multiple Maps are not supported.
@Hypfer : Ticket can be closed as there does not seem to be a Problem with Valetudo in that regard.

closed 2020-10-07 by Hypfer

petec999, 2020-10-07:
@Hypfer I just want make a workaround in my situation, I noticed, that this is not supported, just trying to make my life (not only mine I think) easier. Thanksbin advance...