Hypfer/Valetudo #553: Roborock: Add "Indicator Light" LED disable feature

Hypfer, 2020-07-05:
Controlled by miio command `set_led_status` and fetched with `get_led_status`, a value of `0` means that the indicator LEDs will be disabled if the robot is docked and fully charged.

This is represented by roborock feature flag `119`

Official App screenshot:
[image]

ccoors, 2021-03-23:
Which capability would this be? The `LEDControlCapability` does not seem to be the right one (at least in its current state). Maybe adding flags to it? Or inventing a new `IndicatorLightControlCapability`?

Hypfer, 2021-03-23:
Indeed the `LEDControlCapability` doesn't really make sense how it is now and can probably be removed.

This feature is more like the simple carpet mode toggle and unsurprisingly can also be found right above that in the official app
[image]

We could probably refactor both to a common base class `BooleanSettingsControlCapability`, however given that there is basically no code in all of them, that hierarchy almost feels like a university java project

closed 2021-11-19 by Hypfer