Hypfer/Valetudo #554: Voice pack installation

Digl, 2020-07-08:
Voice Pack upload succeeds but it doesn't change.
Does anybody have the same problem?

Hypfer, 2020-07-25:
This doesn’t sound like a bug in Valetudo. I hope you can understand that we cannot provide individual support for issues not directly related to the Valetudo software.
You can, however, potentially ask for help on [roboter-forum.com](https://www.roboter-forum.com/index.php?thread/29822-valetudo-alternatives-cloudfreies-webinterface-ohne-dustcloud-mit-livemap-auf-de/) (since the forum is in German you may want to use a translation tool before posting your question there).

closed 2020-07-25 by Hypfer

Digl, 2020-07-25:
No Problem! I think the voice pack itself was corrupt.
Thank you!