Hypfer/Valetudo #563: WiFi disconnects after enabling MQTT with default settings

ne0phyte, 2020-07-19:
## Describe the bug

Opening the MQTT settings, enabling MQTT, not changing any other configuration value and saving that, the robot started disconnecting from WiFi in 20s intervals (with 1-3s inbetween while it reconnected).

The web interface would eventually load but by the time I tried to open any dialog or save any settings it was already not responding to requests anymore or the WiFi connection was completely gone again.

## To Reproduce

1. Open web interface
2. Go to Settings
3. Go to MQTT
4. Check "Enabled" checkbox [X] 
5. Click "Save MQTT configuration"
6. Wait a few seconds

### Screenshots

### Vacuum Model

S5

### Valetudo Version

Tested on 0.5.2, 0.5.3

### Firmware Version

2020, no reset fix

### User-Agent

Doesn't matter in this case

## Expected behavior

Valetudo should not crash, WiFi should not disconnect.

## Additional context

Looks like the example mqtt server host is the cause:

    [2020-07-19T12:10:44.845Z] [ERROR] MQTT error: Error: getaddrinfo ENOTFOUND foobar2.example

Since disabling via webinterface is quite tricky once it's in the reconnect/disconnect tango, you can use this (replace ip) to sent the required rest request the second it connects to wifi again:
    
    curl 'http://192.168.178.50/api/mqtt_config' -X 'PUT' -H 'Content-Type: application/json' \                                                                                                      
    --data-binary '{"enabled":'$enabled',"server":"foobar2.example","port":"1883","clientId":"","username":"","password":"","usetls":false,"caPath":"","qos":0,"identifier":"rockrobo","topicPre fix":"valetudo","autoconfPrefix":"","provideMapData":false}'

Hypfer, 2020-07-20:
https://github.com/Hypfer/Valetudo/blob/51a3564c8a1321eab400031a219b74df3394f2ae/lib/MqttClient.js#L209-L215

Why is this segfaulting 🤨

Hypfer, 2020-07-20:
Okay I've been able to reproduce this. The interesting observation is, that it's not enough to stop/restart Valetudo. Even with it stopped, the robot will continue this endless cycle until it is rebooted:

```
Jul 20 09:25:48 home dnsmasq-dhcp[31893]: DHCPDISCOVER(enp3s0) 78:11:dc:ff:ff:ff
Jul 20 09:25:48 home dnsmasq-dhcp[31893]: DHCPOFFER(enp3s0) 10.1.2.3 78:11:dc:ff:ff:ff
Jul 20 09:25:48 home dnsmasq-dhcp[31893]: DHCPREQUEST(enp3s0) 10.1.2.3 78:11:dc:ff:ff:ff
Jul 20 09:25:48 home dnsmasq-dhcp[31893]: DHCPACK(enp3s0) 10.1.2.3 78:11:dc:ff:ff:ff
Jul 20 09:26:08 home dnsmasq-dhcp[31893]: DHCPDISCOVER(enp3s0) 78:11:dc:ff:ff:ff
Jul 20 09:26:08 home dnsmasq-dhcp[31893]: DHCPOFFER(enp3s0) 10.1.2.3 78:11:dc:ff:ff:ff
Jul 20 09:26:08 home dnsmasq-dhcp[31893]: DHCPREQUEST(enp3s0) 10.1.2.3 78:11:dc:ff:ff:ff
Jul 20 09:26:08 home dnsmasq-dhcp[31893]: DHCPACK(enp3s0) 10.1.2.3 78:11:dc:ff:ff:ff
Jul 20 09:26:28 home dnsmasq-dhcp[31893]: DHCPDISCOVER(enp3s0) 78:11:dc:ff:ff:ff
Jul 20 09:26:28 home dnsmasq-dhcp[31893]: DHCPOFFER(enp3s0) 10.1.2.3 78:11:dc:ff:ff:ff
Jul 20 09:26:28 home dnsmasq-dhcp[31893]: DHCPREQUEST(enp3s0) 10.1.2.3 78:11:dc:ff:ff:ff
Jul 20 09:26:28 home dnsmasq-dhcp[31893]: DHCPACK(enp3s0) 10.1.2.3 78:11:dc:ff:ff:ff
Jul 20 09:26:48 home dnsmasq-dhcp[31893]: DHCPDISCOVER(enp3s0) 78:11:dc:ff:ff:ff
```

Only difference here being that it stays responsive for the most part. I'd guess that's because of the router though.

commit 38a0af82 (2020-07-20): fix(mqtt): Check DNS resolution before connecting to mqtt broker to fix #563
closed 2020-07-20 by Hypfer