Hypfer/Valetudo #565: Stuck on loading after enabling http authentication

yieldhog, 2020-07-19:
## After reboot, pages get stuck on loading (particularly config options, will not load.

**This issue appears to have occurred after I enabled http authentication**

### Screenshots
[image]

### Vacuum Model
S5 Gen 2

### Valetudo Version
0.5.3

### Firmware Version
3.5.8 Build 002020

### User-Agent
Safari, MacOS / iOS

## Additional context
I am able to control vacuum and connect via rcc -- however, pages keep freezing up, even after reboot. Had been working before I enabled http authentication. I can connect via ssh, is there a way to disable http authentication via ssh?

Also, I noticed when trying to follow troubleshooting docs, the command "service" is unavailable for me via ssh. I have never updated the stock firmware, but followed installation instructions and used dust builder for firmware.

Thanks

yieldhog, 2020-07-19:
Appears to have been an issue with Safari (or an extension I have installed?).. worked perfectly on Chrome. I disabled http authentication via Chrome browser and it works fine again now on Safari.

Hypfer, 2020-07-20:
#312 is possible related as well.

Since I do not own any apple devices and apple refuses to provide safari for any other platform, I'm afraid that I can't fix this. :(

closed 2020-08-06 by Hypfer

yieldhog, 2020-08-06:
Will test as well, thanks much for the fix!

commit 252c22dc (2020-08-06): fix(ui): fix #565 ui not working with safari and basic auth