Hypfer/Valetudo #568: MQTT Current Version

fperezm, 2020-07-27:
Hi,
I would like to create an automation in Homeassistant so that it alerts me when the version I have installed does not match the last one published.
In the integration with MQTT this data does not seem, is there a possibility to include it?

Thank you very much in advance

Hypfer, 2020-07-27:
Interesting idea.

Might be something for #542

Hypfer, 2020-07-30:
Fixed in 0.6.1 which will be released in a few minutes.

It would be great if you could share your automation by adding it to the docs here: https://github.com/Hypfer/Valetudo/blob/master/docs/_pages/integrations/home-assistant-integration.md

```
sensor:
  - platform: rest
    resource: https://api.github.com/repos/Hypfer/Valetudo/releases
    name: "Current Valetudo Version"
    value_template: '{{ value_json[0].tag_name }}'
```

This is one part of it, but I have no idea how to access the sw_version property :(

commit a23d5af9 (2020-07-30): feat(mqtt): Added sw_version to mqtt autodiscovery to fix #568
closed 2020-07-30 by Hypfer