Hypfer/Valetudo #569: Segmentation fault with MQTT enabled

DomiStyle, 2020-07-28:
## Describe the bug
Starting with 0.6.0 Valetudo crashes on startup when MQTT is enabled, after some time it comes back online but with MQTT disabled.

## To Reproduce
1. Update to 0.6.0
2. Enable MQTT via web interface
3. Valetudo crashes and keeps restarting a few times

### Logs
The logs right after enabling from the web interface and after the first restart:
```
Segmentation fault (core dumped)
[2020-07-28T18:52:37.114Z] [INFO] Loading configuration file: /mnt/data/valetudo/config.json
[2020-07-28T18:52:37.138Z] [INFO] Starting Valetudo for Vacuum Model roborock.vacuum.s6
[2020-07-28T18:52:37.139Z] [INFO] DeviceId 1234
[2020-07-28T18:52:37.139Z] [INFO] IP 127.0.0.1
[2020-07-28T18:52:37.140Z] [INFO] CloudSecret abcd
[2020-07-28T18:52:37.142Z] [INFO] LocalSecret abcd
[2020-07-28T18:52:37.143Z] [INFO] JS Runtime Version v14.4.0
[2020-07-28T18:52:37.144Z] [INFO] Max Heap Size: 33.5 MiB
[2020-07-28T18:52:37.254Z] [DEBUG] >>> local: HandshakePacket()
[2020-07-28T18:52:37.362Z] [INFO] Dummycloud is spoofing 203.0.113.1:8053 on 127.0.0.1:8053
[2020-07-28T18:52:37.366Z] [INFO] Webserver running on port 80
Segmentation fault (core dumped)
```

### Vacuum Model
Roborock S6

### Valetudo Version
0.6.0

### Firmware Version
1998

## Expected behavior
MQTT server is started

## Used config
```
"mqtt": {
    "enabled": true,
    "server": "mqtt.example.org",
    "port": "1883",
    "clientId": "",
    "username": "",
    "password": "",
    "usetls": false,
    "ca": "",
    "clientCert": "",
    "clientKey": "",
    "qos": 0,
    "identifier": "vacuum",
    "topicPrefix": "valetudo",
    "autoconfPrefix": "homeassistant",
    "provideMapData": false
  }
```

Hypfer, 2020-07-28:
Yep. Thats caused by dns.lookup() failing with the statically linked nodejs base binary.

I've added a hackish mitigation with 41f3e98

You might want to try the latest master

DomiStyle, 2020-07-28:
Yeah, that seems spot on. Entering the IP works perfectly.

On a related note, the checkboxes on the MQTT page do not reflect the active config. MQTT is enabled in the config.json but shown as off in the UI. This is true for all the checkboxes (Enabled, SSL/TLS, Provide map data).

Hypfer, 2020-07-28:
Fixed the latter with 039b7411

closed 2020-07-28 by Hypfer