Hypfer/Valetudo #570: Allow manual room splitting and merging

DomiStyle, 2020-07-28:
Since 0.6.0 Valetudo provides room support (which works great!).

However, the Roborock and Xiaomi apps both provide features to split/merge rooms manually in case the automatic room detection doesn't properly detect rooms (as it usually does for me).

Even a command line tool/option to specify coordinates at which to split or zones to merge would be fine with me since it's only necessary once at vacuum setup. Integration into the map UI would be perfect.

rumpeltux, 2020-08-17:
Here are a few commands I once captured around these features for the viomi-v7.
```
{'from': '4', 'id': 9558, 'method': 'arrange_room', 'params': {'lang': 'de', 'mapID': 1577967823, 'pointArr': [[1, '3.8_0', '5.4_2.35']], 'roomID': 11, 'type': 1}}
{'from': '4', 'id': 9566, 'method': 'arrange_room', 'params': {'lang': 'de', 'mapID': 1577967823, 'roomArr': [[10, 11]], 'type': 0}}
{'from': '4', 'id': 9570, 'method': 'rename_room', 'params': [1577967823, 1, 11, 'Wohnzimmer']}
```

10 and 11 are room ids which should be persisted in your valetudo.conf. Request 9558 is a split across line coords in pointArr (unit is roughly meters IIRC), 9566 is a merge.

You can probably use the viomivacuum raw_command from [miio-cli](https://github.com/rytilahti/python-miio) to issue the commands to the robot.

Would also love to see this integrated in the map UI.

DomiStyle, 2020-08-17:
> You can probably use the viomivacuum raw_command from [miio-cli](https://github.com/rytilahti/python-miio) to issue the commands to the robot.

Can you even use miio-cli to send cloud commands to the vacuum? Does Valetudo have a way of sending manual commands?

Hypfer, 2020-08-17:
>10 and 11 are room ids which should be persisted in your valetudo.conf.

Why is that? Afaik those are part of the map data and therefore don't need to be stored at all

Hypfer, 2020-08-17:
The framework to implement this will be part of #584

This is the Capability that will need to be implemented for each vendor:
https://github.com/Hypfer/Valetudo/blob/e125dabd27b7392bafd5ccba8dab5adad5a9b72e/lib/core/capabilities/MapSegmentationCapability.js#L4-L43

rumpeltux, 2020-08-17:
> > 10 and 11 are room ids which should be persisted in your valetudo.conf.
> 
> Why is that? Afaik those are part of the map data and therefore don't need to be stored at all

Technically yes, but in the existing codebase all rooms were in the config object and maps weren’t kept in memory.
So to make the map rooms available to UI/commands I persisted them in the config object too.

> Can you even use miio-cli to send cloud commands to the vacuum? Does Valetudo have a way of sending manual commands?

Yes, most commands are accepted on the local interface, except most notably the OTA / firmware update ones.
Valetudo doesn’t have a way to do this, but a /raw_cmd http handler should be almost trivial to add and is probably useful for other development purposes too (e.g. starting OTA updates).

DomiStyle, 2020-08-17:
> Yes, most commands are accepted on the local interface, except most notably the OTA / firmware update ones.

In that case the Roborock S6 probably has a different syntax, ``arrange_room`` returns ``unknown_method``.

Hypfer, 2020-08-17:
Yes the commands are vendor-specific and sometimes even robot-specific.

For the S5 and S6 they are `split_segment` and `merge_segment`

DomiStyle, 2020-08-18:
> For the S5 and S6 they are split_segment and merge_segment

Do you have the full message payload? Would love to try and split the rooms manually until this feature is implemented.

Hypfer, 2020-08-18:
https://github.com/rand256/valetudo/blob/ab9fe2fb379f7f10d9a1556739b4fcc78cbfe03b/lib/miio/Vacuum.js#L929-L951

see here

DomiStyle, 2020-08-24:
@Knapoc Yes, I did it the same way except reading the coordinates from the developer console of the browser.

The segment id can also be read this way, just start a cleanup of a segment and read the id from the developer console.

Hypfer, 2020-08-24:
The segment id can also be read by zooming in to a certain level. Then, they will be drawn on the map as seen in the 0.6.0 changelog

DomiStyle, 2020-08-24:
> The segment id can also be read by zooming in to a certain level.

Ah, that makes things a little bit faster.

The points also need to be far enough apart in the zone, splitting just inside a door frame didn't work for me.

Hypfer, 2021-01-09:
The backend code is there now. We just need UI support

Knapoc, 2021-03-11:
> [e306569](https://github.com/Hypfer/Valetudo/commit/e306569d2daaa6f8ab7cfe75d3cd4a99b37faa8b)
> 
> Testing much appreciated

just compiled the master branch, tested merging & splitting on my S6 and it's working great. Awesome work - thank you!

Hypfer, 2021-03-15:
https://github.com/Hypfer/Valetudo/releases/tag/2021.03.0

closed 2021-03-15 by Hypfer