Hypfer/Valetudo #572: MQTT with hostname not working

darootler, 2020-07-29:
I have the following configuration working:

>  "mqtt": {
>     "enabled": true,
>     "server": "192.168.X.X",
>     "port": 1883,
>     "clientId": "",
>     "username": "myuser",
>     "password": "myasswd",
>     "usetls": false,
>     "caPath": "/mnt/data/valetudo/ca.pem",
>     "qos": 0,
>     "identifier": "abstellraum-staubsauger",
>     "topicPrefix": "valetudo",
>     "autoconfPrefix": "homeassistant",
>     "provideMapData": true
>   },

This configuration is not working:

>  "mqtt": {
>     "enabled": true,
>     "server": "mqtthost.my.domain",
>     "port": 1883,
>     "clientId": "",
>     "username": "myuser",
>     "password": "myasswd",
>     "usetls": false,
>     "caPath": "/mnt/data/valetudo/ca.pem",
>     "qos": 0,
>     "identifier": "abstellraum-staubsauger",
>     "topicPrefix": "valetudo",
>     "autoconfPrefix": "homeassistant",
>     "provideMapData": true
>   },

Can someone point in the right direction how to get this working with the hostname set? I am able to ping the hostname from the vacuum command line.

Regards
Richard

Poeschl, 2020-07-29:
Having the same. Valetudo 0.6.0 only starts up fully when the broker host is specified as ip

When setting the broker with hostname like:
```
 "mqtt": {
    "enabled": true,
    "server": "p-assist.fritz.box",
    "port": "1883",
    "clientId": "",
    "username": "roborock",
    "password": "*****",
    "usetls": false,
    "ca": "",
    "clientCert": "",
    "clientKey": "",
    "qos": 0,
    "identifier": "P-Dusty",
    "topicPrefix": "valetudo",
    "autoconfPrefix": "homeassistant",
    "provideMapData": true
  },
```

I'm getting a segmentation fault at startup:
```
[2020-07-29T16:14:40.975Z] [INFO] Loading configuration file: /mnt/data/valetudo/config.json
[2020-07-29T16:14:41.005Z] [INFO] Starting Valetudo for Vacuum Model roborock.vacuum.s5
[2020-07-29T16:14:41.007Z] [INFO] DeviceId 117807017
[2020-07-29T16:14:41.008Z] [INFO] IP 127.0.0.1
[2020-07-29T16:14:41.009Z] [INFO] CloudSecret qoNw0gnTDa4qgTSf
[2020-07-29T16:14:41.013Z] [INFO] LocalSecret dBom8LolaCVimE1C
[2020-07-29T16:14:41.013Z] [INFO] JS Runtime Version v14.4.0
[2020-07-29T16:14:41.015Z] [INFO] Max Heap Size: 33.5 MiB
[2020-07-29T16:14:41.232Z] [INFO] Dummycloud is spoofing 203.0.113.1:8053 on 127.0.0.1:8053
[2020-07-29T16:14:41.237Z] [INFO] Webserver running on port 80
Segmentation fault (core dumped)
```

darootler, 2020-07-29:
Good to hear that this is not a problem related to my environment. 

But this, for sure, breaks MQTTS configurations.

Regards
Richard

Hypfer, 2020-07-29:
Duplicate of #569 569

closed 2020-07-29 by Hypfer