Hypfer/Valetudo #573: Many iterations

tremebundo, 2020-07-29:
Hi,

Sorry if this is not a bug but i could not find where to set the number of iterations for a cleaning.
I have just installed Valetudo 0.6, did a first running at the complete home, defined a zone and when I order the vacuum to clean that zone it takes for ever to finish.
Certanly it will finish but I could not count how many iterations it dit at that zone. Lots. And it takes hours to finish.
Am I doing something wrong?

Thanks in advance.

Hypfer, 2020-07-29:
I've also experienced this. It's most likely not a bug in valetudo but rather in the roborock firmware

Try with a smaller zone. It might be possible that there are some parts of the zone out of reach for the robot which forces him to endlessly trying to reach it

maikkoe, 2020-07-30:
I have the same problem, it doesn't matter where or how big the zone is. For testing, I have created a very small zone in the middle of the room. I did not count how many times he cleaned the zone, but felt it ~10 times. I have tried different firmware versions (v11_004007 & 11_004018), the problem only exists with 0.6.0.

closed 2020-07-30 by Hypfer

alex1702, 2020-07-30:
Thanks. That was really confusing and an unusable function ^^

commit f3cc9354 (2020-07-30): fix(vacuum.roborock): Don't multiply iterations by 10 to fix #573