Hypfer/Valetudo #575: MQTT over TLS

darootler, 2020-07-30:
## Describe the bug
I tried to connect the vacuum's mqtt client over tls without success. I already managed to get a test working with the following:

`mosquitto_pub -h my.host.name -p 8883 -u myuser -p mypwd -t test -m test --cafile /ssl/ca.pem`

Here is my config:

```
{
  "spots": [],
  "mqtt": {
    "enabled": true,
    "server": "my.host.name",
    "port": 8883,
    "clientId": "",
    "username": "myuser",
    "password": "mypwd",
    "usetls": true,
    "ca": "/mnt/data/valetudo/ca.pem",
    "qos": 0,
    "identifier": "abstellraum-staubsauger",
    "topicPrefix": "valetudo",
    "autoconfPrefix": "homeassistant",
    "provideMapData": true
  },
```

This is the error the vacuum logs:

`[2020-07-30T13:43:48.425Z] [ERROR] MQTT error: Error: unable to verify the first certificate`

This is the error mosquitto logs:

```
1596116619: New connection from 192.168.5.3 on port 8883.
1596116620: Socket error on client <unknown>, disconnecting.
```

### Vacuum Model
Gen1

### Valetudo Version
0.6.1
### Firmware Version
4018

### User-Agent
Chrome/75.0.3770.142

## Expected behavior
MQTT Client connects to the broker using TLS

Hypfer, 2020-07-30:
>"ca": "/mnt/data/valetudo/ca.pem",

Valetudo doesn't expect a path there but rather the certificate as a string.

[image]

That's why these are textboxes

darootler, 2020-07-31:
Thanks, it's working now. That wasn't clear to me, is that documented anywhere?

Regards
Richard

closed 2020-07-31 by darootler

Hypfer, 2020-07-31:
Not yet.

Feel free to extend the documentation by clicking on the edit symbol on the top-right corner:
https://github.com/Hypfer/Valetudo/blob/master/docs/_pages/integrations/mqtt.md

Hypfer, 2020-07-31:
The reasoning behind this is, that it should be possible to configure this without requiring ssh access and building a file upload is hard.

darootler, 2020-07-31:
> The reasoning behind this is, that it should be possible to configure this without requiring ssh access and building a file upload is hard.

Very good idea.

Regards
Richard