Hypfer/Valetudo #576: Zones clean more than once

loopphoto, 2020-08-01:
## Describe the bug

In 0.60, zones clean 3 times before proceeding to the dock or next zone.
## To Reproduce

Steps to reproduce the behavior:
Choose a zone to clean, or manually draw zone then clean.
### Screenshots

[image]

### Vacuum Model

Roborock gen 1 - Xiaomi Mi Robot Vacuu,
### Valetudo Version

0.60
### Firmware Version

4018
### User-Agent

Chrome
## Expected behavior

Select zone and clean zone once. It would be able to choose how many times to clean like the Mi app, but defaulting to 3x is not preferred. Previous version of Valetudo defaulted to 1.
## Additional context

Thanks very much for the great work!

maikkoe, 2020-08-02:
Fixed in 0.61 see #573

closed 2020-08-02 by Hypfer

loopphoto, 2020-08-02:
Thanks a lot!
--
Regards

*Preyen Moodley*

*Photographer*

--
*instagram* <http://instagram.com/loopphoto>

*website* <http://www.loopphoto.co.za/>

*facebook* <http://www.facebook.com/preyenmoodley>

*preyen@loopphoto.co.za* <preyen@loopphoto.co.za>

*+27 83 643 4527* <+27%2083%20643%204527>

--

DISCLAIMER: This document is for the personal and private attention of the
addressee and should be read by the addressee only.

Loop Photography C.C. its subsidiaries and/or any of its associate
companies accepts no liability and/or responsibility whatsoever for all and
any consequences arising out of but not limited to the receipt by a party,
other than the addressee, of this communication. If you have received this
communication in error, please notify us immediately thereof by telephone.
Kindly destroy this communication immediately. Any form of reproduction,
dissemination, copying, disclosure, modification, distribution and/or
publication of this communication is strictly prohibited.