Hypfer/Valetudo #583: "Unknown cloud message received" for Viomi v8.

PaskoMoto, 2020-08-10:
Hi.

Checcking the logs of my viomi v8 I see a pair of methods not recognized (by DummyCloud, I suppose): 
"_otc.ncinfo" and "_otc.ncstat"

I guess they could provide info about use and consummables status.... 

```
Aug 10 07:24:37 aspirador.lan valetudo[2311]: [2020-08-10T05:24:37.970Z] [INFO] Unknown cloud message received: {"id":1427363063,"method":"_otc.ncinfo","params":{"miio_ver":"0.0.5","miio_times":[35073,-1,24609,10355],"ot_stat":[3,2,3],"otu_stat":[3,0,80,114,4,3,635,511,5],"ott_stat":[0,0,-1,-1,-1,-1,0,0,0]}}

Aug 10 07:24:37 aspirador.lan valetudo[2311]: [2020-08-10T05:24:37.974Z] [INFO] Unknown cloud message received: {"method":"_otc.ncstat","id":321282685,"params":{"timestamp":1597011015,"proto":0,"otip":"110.43.0.83","otport":8053,"sendcnt":95,"ackcnt":75,"ackcnt_timeout":0,"costtime":316,"rpc":38,"rpcack":44,"rpc_local":0,"rpcack_local":0,"dns":2,"keepalive_interval":20}}

```

How easy would it be to add these methods so that they are supported? I'd try , but I've read the code and need a some guidance! :D :D

Best Regards.
PaskoMoto.

Hypfer, 2020-08-10:
I'd strongly advice against adding that feature right now since I'm currently reworking the internals on that.
It should get much easier (and especially easier to understand) when that is done. No ETA though

PaskoMoto, 2020-08-10:
Hi.

Ok. I agree with your suggestion. ;)

Best Regards.
PaskoMoto.

rumpeltux, 2020-08-17:
I thought of these more like network and protocol stats. Maybe I’m wrong, but I don’t think those messages provide any value right now.
There are separate APIs to collect consumable information. So for these two cloud messages should probably just implement a no-op handler to avoid them spamming logs and confusing users.

Hypfer, 2021-01-09:
The new system has been merged to master. You can go ahead now :)

rumpeltux, 2021-01-31:
I’m closing due to lack of activity. Feel free to reopen a PR if you end up implementing this (though I see little or no value at this point).

closed 2021-01-31 by rumpeltux

forast3r, 2021-04-13:
Still appearing in the logs:

```
[2021-04-11T19:37:55.069Z] [INFO] Unknown cloud message received: {"id":685065857,"method":"_otc.ncinfo","params":{"miio_ver":"0.0.5","miio_times":[70212,-1,0,70206],"ot_stat":[1,0,1],"otu_stat":[1,0,42,142,12,14,281,197,84],"ott_stat":[0,0,-1,-1,-1,-1,0,0,0]}}
[2021-04-11T19:37:55.090Z] [INFO] Unknown cloud message received: {"method":"_otc.ncstat","id":2074490832,"params":{"timestamp":1618168075,"proto":0,"otip":"127.0.0.1","otport":8053,"sendcnt":5,"ackcnt":5,"ackcnt_timeout":0,"costtime":47,"rpc":0,"rpcack":0,"rpc_local":0,"rpcack_local":0,"dns":2,"keepalive_interval":20}}
```