Hypfer/Valetudo #587: Support Dreame F9 vacuum

urbatecte, 2020-08-18:
## Describe the bug
Dreame F9 vacuum get lauched beginning of this year.
Could be nice to be supported in Valetudo ;o)

## To Reproduce

### Screenshots

### Vacuum Model

dreame.vacuum.p2008

### Valetudo Version

### Firmware Version

### User-Agent

## Expected behavior

## Additional context

If you need other data, as far as I can do it, I'll be happy to help.

Hypfer, 2020-08-18:
Why is it so hard for people to use the issue templates properly? I just don't understand.

urbatecte, 2020-08-18:
You may laught at me if I tell you that I carrefully read the template and was a bit confused regarding my request.
Sorry for that.

Hypfer, 2020-08-18:
Looks like they've changed a lot for the F9 so it won't be possible to simply take the 1C implementation and be done with it

urbatecte, 2020-08-28:
Ok, should I keep the PR open just in case ?
Otherwise I'll close it.
Thanks for your work !

rumpeltux, 2020-08-28:
Can you link the PR here? I didn’t see it.

urbatecte, 2020-08-28:
Sorry my bad, due to proffessionnal habit (problem report) I mean do you want me to let the issue open ?

rumpeltux, 2020-08-28:
We’ll need a developer owning such a device to implement the core functionality and open a respective Pull Request.
Closing until then.

bigdx, 2020-11-25:
Is there any progress? As Dennis website states rooting is possible and a firmware (valetudo support) avialable... but, I cant find any files or further informations :-/

Hypfer, 2020-11-25:
As of now, there's no public root available and neither has any work happend on supporting the F9 in Valetudo.

What has been done however is work to make it easier to support new Vacuums so when that is finished, it should also be much easier to add support for the F9.

There aren't any ETAs for either though

bigdx, 2020-11-25:
> As of now, there's no public root available and neither has any work happend on supporting the F9 in Valetudo.
>
> What has been done however is work to make it easier to support new Vacuums so when [...]

Ah ok ... Maybe I was a little to enthusiastic after I have read "yes" next to rootable and valetudo ;-)
I guess the same applies to the S5 Max, right?

Hypfer, 2020-11-25:
Yes

bigdx, 2020-11-25:
Ahm, and Xiaomi MiJia 1C too? Cant find files or installation- or rooting-instructions ...

Hypfer, 2020-11-25:
Yes. And that's definitely nothing that belongs in this issue
Honestly, you could've deduced that by thinking about the previous two questions.

Hypfer, 2021-01-09:
The miot spec looks pretty much identical to the D9, for which support was added by me very recently.

_If_ the map format behaves the same as on the D9, this could be a very easy addition maybe very soon

Hypfer, 2021-02-02:
The map format _is_ the same as on the D9 (and the 1C for that matter).

It's basically the same robot without Lidar

Hypfer, 2021-03-23:
4fe0ff7fd952533a073cd51432626f99aeadd856

Untested, but we'll find out soon

closed 2021-03-23 by Hypfer

maluueu, 2021-04-10:
Anything new ?